MichaIng/DietPi #1077: [RPi Stretch] Stretch distribution available on "archive.raspberrypi.org" repository

MichaIng, 2017-07-15:
While checking out the need of package `apt-transport-https` and which repos are actually available with https access, I accidently found `archive.raspberrypi.org` not only available by https, but also finally offer the long awaited stretch distro: https://archive.raspberrypi.org/debian/dists/stretch/

I directly changed my `/etc/apt/sources.list.d/raspi.list` accordingly and successfully did an `apt update`, via https by the way, followed by some package upgrades from that repo 😃 :

```
2017-07-16 00:13:06 root@micha:/var/log# apt update
Get:1 https://archive.raspberrypi.org/debian stretch InRelease [17.7 kB]
Get:2 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch InRelease [15.0 kB]
Get:3 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 https://archive.raspberrypi.org/debian stretch/main armhf Packages [80.4 kB]
Get:5 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [21.6 kB]
Get:6 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/contrib armhf Packages [56.8 kB]
Get:7 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/non-free armhf Packages [95.2 kB]
Get:8 https://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/rpi armhf Packages [1,360 B]
Fetched 12.0 MB in 19s (619 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
9 packages can be upgraded. Run 'apt list --upgradable' to see them.
2017-07-16 00:13:41 root@micha:/var/log# apt list --upgradable
Listing... Done
device-tree-compiler/testing 1.4.4-1 armhf [upgradable from: 1.4.2-1]
libpam-modules/testing 1.1.8-3.6+rpi1 armhf [upgradable from: 1.1.8-3.6]
libpam-modules-bin/testing 1.1.8-3.6+rpi1 armhf [upgradable from: 1.1.8-3.6]
libpam-runtime/testing 1.1.8-3.6+rpi1 all [upgradable from: 1.1.8-3.6]
libpam0g/testing 1.1.8-3.6+rpi1 armhf [upgradable from: 1.1.8-3.6]
libraspberrypi-bin/testing 1.20170703-2 armhf [upgradable from: 1.20161215-1]
libraspberrypi0/testing 1.20170703-2 armhf [upgradable from: 1.20161215-1]
raspberrypi-bootloader/testing 1.20170703-2 armhf [upgradable from: 1.20161215-1]
raspberrypi-kernel/testing 1.20170703-2 armhf [upgradable from: 1.20161215-1]
2017-07-16 00:14:24 root@micha:/var/log# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
The following packages will be upgraded:
  device-tree-compiler libpam-modules libpam-modules-bin libpam-runtime libpam0g
5 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Need to get 1,072 kB of archives.
After this operation, 9,216 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 https://archive.raspberrypi.org/debian stretch/main armhf libpam0g armhf 1.1.8-3.6+rpi1 [120 kB]
Get:2 https://archive.raspberrypi.org/debian stretch/main armhf libpam-modules-bin armhf 1.1.8-3.6+rpi1 [102 kB]
Get:3 https://archive.raspberrypi.org/debian stretch/main armhf libpam-modules armhf 1.1.8-3.6+rpi1 [289 kB]
Get:4 https://archive.raspberrypi.org/debian stretch/main armhf libpam-runtime all 1.1.8-3.6+rpi1 [212 kB]
Get:5 https://archive.raspberrypi.org/debian stretch/main armhf device-tree-compiler armhf 1.4.4-1 [349 kB]
Fetched 1,072 kB in 4s (233 kB/s)
Preconfiguring packages ...
(Reading database ... 21302 files and directories currently installed.)
Preparing to unpack .../libpam0g_1.1.8-3.6+rpi1_armhf.deb ...
Unpacking libpam0g:armhf (1.1.8-3.6+rpi1) over (1.1.8-3.6) ...
Setting up libpam0g:armhf (1.1.8-3.6+rpi1) ...
(Reading database ... 21302 files and directories currently installed.)
Preparing to unpack .../libpam-modules-bin_1.1.8-3.6+rpi1_armhf.deb ...
Unpacking libpam-modules-bin (1.1.8-3.6+rpi1) over (1.1.8-3.6) ...
Setting up libpam-modules-bin (1.1.8-3.6+rpi1) ...
(Reading database ... 21302 files and directories currently installed.)
Preparing to unpack .../libpam-modules_1.1.8-3.6+rpi1_armhf.deb ...
Unpacking libpam-modules:armhf (1.1.8-3.6+rpi1) over (1.1.8-3.6) ...
Setting up libpam-modules:armhf (1.1.8-3.6+rpi1) ...
(Reading database ... 21302 files and directories currently installed.)
Preparing to unpack .../libpam-runtime_1.1.8-3.6+rpi1_all.deb ...
Unpacking libpam-runtime (1.1.8-3.6+rpi1) over (1.1.8-3.6) ...
Setting up libpam-runtime (1.1.8-3.6+rpi1) ...
(Reading database ... 21302 files and directories currently installed.)
Preparing to unpack .../device-tree-compiler_1.4.4-1_armhf.deb ...
Unpacking device-tree-compiler (1.4.4-1) over (1.4.2-1) ...
Processing triggers for libc-bin (2.24-11+deb9u1) ...
Setting up device-tree-compiler (1.4.4-1) ...
```

Everything runs fine so far. No obsolete/missing package, no broken dependency. At least for the RPi Stretch image it make sense to update the respberrypi repo accordingly. Also I like to access by https, if we already have `apt-transport-https` inside our dependencies list.
Also for the raspbian repo, there are some https mirrors available: http://raspbian.org/RaspbianMirrors
Maybe we could prefer them over http/ftp as standard repo, configurable via dietpi-config for the related countries? https 4tw :)

Fourdee, 2017-07-16:
@MichaIng 

> archive.raspberrypi.org

Great find 👍

We can patch for this in the ```patch_file```.

```
if (( $HW_MODEL < 10 && DISTRO == 4 )); then # RPi + Stretch
fi
```

We'll also need to update https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L80-L83 

> Also for the raspbian repo, there are some https mirrors available: http://raspbian.org/RaspbianMirrors
Maybe we could prefer them over http/ftp as standard repo, configurable via dietpi-config for the related countries? https 4tw :)

Additional entries can be added easily here:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-config#L4637-L4645

@MichaIng 
You up for giving this a commit, or want me to do this one?

MichaIng, 2017-07-16:
I will do a start with the patch_file.
> We'll also need to update https://github.com/Fourdee/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L80-L83

As lines 80 to 83 are about the raspbian repo, do I get it right, that it's actually line 89 where the second comment need to be removed? As Jessie is still the RPi stable repo, I am not sure how you handle the stretch (test) updates and images ;).

About the https thing:
- I would move that to an own pull request/issue. The questions are, if we just want to add some optional https mirrors to dietpi-config or want to stringently switch all repos to https, where possible for security/privacy reasons.
- I mean it's "just" about package updates/upgrades and by default I don't see any https mirror in any distro 😛 .
- Inside the dietpi code I just see one single use of "deb https": https://github.com/Fourdee/DietPi/search?q="deb+https"
and this one is also reachable on http: http://apt.sonarr.tv/
So the alternative would be to just remove the "s" there and remove "apt-transport-https" from the dietpi dependencies in favour of diet to privacy ;).

Fourdee, 2017-07-16:
@MichaIng 

>  will do a start with the patch_file.

Excellent 👍 

> As lines 80 to 83 are about the raspbian repo, do I get it right, that it's actually line 89 where the second comment need to be removed? As Jessie is still the RPi stable repo, I am not sure how you handle the stretch (test) updates and images ;).

I'll do the pre file, its a little messy and needs cleaning up :)

>  don't see any https mirror in any distro 😛 .

We had some users reporting issues with ```https``` connections failing. Hence we geared towards the "fix all" solution with ```http```.

> I would move that to an own pull request/issue. The questions are, if we just want to add some optional https mirrors to dietpi-config or want to stringently switch all repos to https, where possible for security/privacy reasons.

Yep, lets go for replacing with ```https```. Although, i'd like to keep at least 1 entry (duplicate if required) for ```http``` as a fallback.

> apt-transport-https

Still required due to:
https://github.com/Fourdee/DietPi/issues/669#issuecomment-269344844

Fourdee, 2017-07-16:
Notes:
- 🈴 Raspbian mirror director does not support ```https```: https://mirrordirector.raspbian.org
- 🈴 Deb ```Err https://security.debian.org jessie/updates/main armhf Packages
  Failed to connect to security.debian.org port 443: Connection refused```

MichaIng, 2017-07-16:
> 🈴 Deb Err https://security.debian.org jessie/updates/main armhf Packages Failed to connect to security.debian.org port 443: Connection refused

Funny thing, I thought is was just about "this" repo to to use the https package for: https://github.com/Fourdee/DietPi/issues/669#issuecomment-269344844

But yeah I can confirm, that https://mirrordirector.raspbian.org/raspbian/dists/ and https://security.debian.org/debian-security/ are not available where they are via http.

> 🈯️ Rpi https://www.mirrorservice.org/sites/archive.raspbian.org/raspbian

I am not too sure about this solution. Looks little complicated and risky for me to use a 3rd party mirror service again just to have an https access, what you think?

I personally would only use https, where it is already available and not try to force it for every package update/upgrade. For the suggested repos in dietpi-config we could just replace them with https ready ones from here: http://raspbian.org/RaspbianMirrors respectively from debian mirror list
and use `http://mirrordirector.raspbian.org/raspbian/` (respectively debian repo) as standard http fallback/option.

Fourdee, 2017-07-16:
@MichaIng 

> I am not too sure about this solution. Looks little complicated and risky for me to use a 3rd party mirror service again just to have an https access, what you think?

This is the default mirror used in Raspbian image. We changed to oxford as it was unreliable, however, oxford is the same recently:
https://github.com/Fourdee/DietPi/issues/518#issuecomment-315604459

> I personally would only use https, where it is already available and not try to force it for every package update/upgrade. For the suggested repos in dietpi-config we could just replace them with https ready ones from here: http://raspbian.org/RaspbianMirrors respectively from debian mirror list
and use http://mirrordirector.raspbian.org/raspbian/ (respectively debian repo) as standard http fallback/option.

Sounds good 👍

Fourdee, 2017-07-16:
@MichaIng 

> Funny thing, I thought is was just about "this" repo to use the https package for: #669 (comment)

Yep, it is, didn't work without it due to redirect. Strange, even their example sources is set to http.
https://wiki.debian.org/SourcesList#Example_sources.list

MichaIng, 2017-07-16:
They might reverted that change due to the problems e.g. you (and for sure others as well) faced with missing and then via apt unavailable `apt-transport-https` package. Like a cat bite it's own tail 😆 .

Now `security.debian.org/debian-security/dists/` is definitely not reachable via https (at least in browser) but is via http and in browser I also do not face any redirection to https.

But nevertheless for my point of view, changing to https where possible and thus supporting the global switch to encrypted internet traffic is the right way to go.

Fourdee, 2017-07-17:
@MichaIng 

> But nevertheless for my point of view, changing to https where possible and thus supporting the global switch to encrypted internet traffic is the right way to go.

Yep agree 👍 

However, appears Debian requires ```http``` for the current repos (as per the wiki). So we'll leave that as is. 

Outstanding tasks:
- Add ```http``` and ```https``` entries for Raspbian repo in ```dietpi-config```

Fourdee, 2017-07-17:
Completed. Added global RPi mirror as ```https```.

@MichaIng 
Please reopen if needed, or, I missed anything :)

closed 2017-07-17 by Fourdee

WolfganP, 2017-07-28:
Naive question: why all the scripts (ie dietpi-set_software , PREP_SYSTEM_FOR_DIETPI.sh and patch_file) modify both /etc/apt/sources.list (the old method of managing apt sources) and /etc/apt/sources.list.d/raspi.list (the newer and preferred method)?
Having both of them present result in duplicated sources (and double checking in case of using mirrors)
Is there any reason for it?
Thx, Wolf.

MichaIng, 2017-07-28:
@WolfganP 

`/etc/apt/sources.list` is still the current method to manage the basic debian/raspbian repos.
`/etc/apt/sources.list.d/` is the place to add "additional" specific repos, without manipulating the distros default sources.list, e.g. the raspberry pi specific `raspi.list`.

There are no duplicates, as both files contain the related different repo URLs.