MichaIng/DietPi #1080: DietPi-Software | JRiver Seg Fault

Fourdee, 2017-07-16:
>

Fourdee, 2017-07-23:
Confirmed with 2x RPi 3's. Unknown cause at the moment, further testing required.

Nudging to v155 to allow release of v154.

Fourdee, 2017-08-26:
I dont believe this to be a DietPi issue, various clean installs, some will work, others wont. Putting this down to a possible bug in Jriver.

Fourdee, 2018-06-29:
Note to self, 2 install count on survey. Flag for removal?

Fourdee, 2018-07-09:
Marking as flagged for removal due to low install count.

MichaIng, 2018-07-09:
@Fourdee 
Makes sense with our install method. The used repo points to wheezy: http://dist.jriver.com/latest/mediacenter/mediacenter22native.list

There are deb packages to be used with Jessie: https://yabb.jriver.com/interact/index.php/topic,115942.0.html?PHPSESSID=vmvle9q70m4vr5it77qda9o2q4

We could test to use them with Stretch, but there is no official Stretch release.

Fourdee, 2018-07-09:
@MichaIng 

> We could test to use them with Stretch, but there is no official Stretch release.

Yep, the software is somewhat buggy at its core, and, with the low install count, not worth our time I believe.

Fourdee, 2018-07-15:
Done.

closed 2018-07-15 by Fourdee