MichaIng/DietPi #1084: Addition of new software complicated.

sytone, 2017-07-16:
Adding in new software is a complicated process. Have you though of breaking it up?

Suggestion:
Under the DietPi Folder make a software folder. Under that make a folder for each application using the id as the folder name. 

software /
    123 /
        info.sh
        install.sh
        uninstall.sh

info.sh - included by default to make the menu. 
Example:
		index_current=123

				 aSOFTWARE_WHIP_NAME[$index_current]='GNUStep'
				 aSOFTWARE_WHIP_DESC[$index_current]='lightweight based on OpenStep'
			aSOFTWARE_CATEGORY_INDEX[$index_current]=0
					  aSOFTWARE_TYPE[$index_current]=0
			 aSOFTWARE_REQUIRES_ALSA[$index_current]=1
	  aSOFTWARE_REQUIRES_XSERVERXORG[$index_current]=1
			 aSOFTWARE_ONLINEDOC_URL[$index_current]='f=8&t=5&start=60#p2072'

install.sh - Called when that optino is selected for a install. Included using . $(dirname "$0")/software/123/install.sh

etc..

Fourdee, 2017-07-17:
> Addition of new software complicated.

DietPi-Software uses arrays at its core. Its a highly efficient system (especially in bash). I find it extremely easy to work with (coming from a C++ background). But yes, I can understand how this may be tricky for some.

We do offer a guide for adding software to DietPi-Software, which covers the steps needed:
https://github.com/Fourdee/DietPi/issues/490#issuecomment-244416570

> Suggestion:
Under the DietPi Folder make a software folder. Under that make a folder for each application using the id as the folder name.
software /
123 /
info.sh
install.sh
uninstall.sh

The reason I decided to keep everything in 1 script? Filesystem is in 4K blocks, so each additional file will use a minimum of 4K, even if its 1byte.
As the entire ```/DietPi``` folder is mounted to RAM, lots of additional "little files" under 4K will increase memory consumption. We wanted to avoid this to keep to our "lightweight" goal.

DietPi-Software, as it stands is too late to change overnight. It would require a near complete re-write, many many hours of testing and I simply don't have the time available at the moment.

Fourdee, 2017-07-19:
I'll mark this as closed.

closed 2017-07-19 by Fourdee