MichaIng/DietPi #1085: Creating a compressed backup image of the OS

vamichael, 2017-07-18:
Can you add [piclone](https://github.com/raspberrypi-ui/piclone) to the software list?  It creates a compressed image of the OS.  In the interim, does anyone know how to do this using `dd`?

I'm kinda bummed because I made a backup using win32diskimager and for some reason the backup was larger than the "disc" and I was unable to restore after I messed up my install, thinking I was covered.

Invictaz, 2017-07-18:
Interesting, the Win32 image is bigger than the actual drive because of compression of some parts.
But PiClone doesn't work on the other SoC's dietpi is running on?

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

ilyasst, 2019-02-16:
Hi, I am very interested by this. It would be great to have an easy way to backup and compress the image. Otherwise, when I back up my dietpi image using Win32DiskImager, the size of the image is as large as the SD card itself. Thanks !

MichaIng, 2019-02-16:
@ilyasst 
Just to clarify some things, that I guess are not clear to everyone:
- PiClone does not produce a compressed image, but an uncompressed clone of the the SDcard to a target drive. Correct me, if I am wrong, but from the scripts source I can't find any compression step. As well this would not make any sense, since it indeed creates partition by partition from `mmcblk0` and copies over the files. So it is no "image" at all, but yeah, a "cloned" drive: https://github.com/raspberrypi-ui/piclone/blob/master/src/backup
- As well it is no real clone. A real clone would be if you use `dd` on an unmounted drive from an external system, so the partition and file system structure would be cloned, instead of the files contained.
- And, if you run this from the running system, booted from `mmcblk0` (SDcard), as well you could have some inconsistencies if the system meanwhile edits files in one place which is already copied and related on another place which is still about to be copied.
- So the final result is something that is close to a clone but besides the partition structure the same than what `dietpi-backup` produces with the same limitations, as mentioned above and no compression.
- You could however create an image file of the generated drive and compress this.
- You could as well compress the backup directory that `dietpi-backup` produces. We though about doing this internally, but after some discussion we decided against this, for the following reasons:
  - Compressing and decompression on every backup run or restore is a quite heavy task that takes time and consumes much system resources.
  - While compressing, on the target drive ~doubled disk space is required, once for the uncompressed files, once for the compressed archive, and potentially even more for temporary files, depending on the compression method. So you need an even larger target drive than without compression.
  - Compressing and decompressing results in lots of disk writes. Not only ~doubled drive space is required, but also doubled disk writes, reducing the drives life time.
  - And even worse, it renders the delta sync useless that we achieve via `rsync` to only copy/edit changed files, instead of the whole system on every backup run. As usually only a small part of the whole system changes during regular backup schedule, this means that the disk writes are in most cases orders of magnitude larger than without compression.

So finally `dietpi-backup` already does a great job as minimal system backup solution.
However it is always advised to as well create a real drive clone by times. And this indeed need to be done from an external system. Win32DiskImager should do the job, or on a Linux system: `dd if=/dev/mmcblk0 of=/path/to/SDCardBackup.img`

### Why is the `img` created larger than the actual drive space consumed on the running system?
The reason is that usually when removing files, the content is deallocated but not really freed, meaning overwritten with zeros. So when cloning the drive/partition with Win32DiskImager or `dd`, this unallocated content is cloned as well. On Linux system you can free unallocated drive space with [zerofree](https://manpages.debian.org/stretch/zerofree/zerofree.8.en.html). Not sure if there is an alternative available on Windows, since since has not `ext` file system support 🤔.

ilyasst, 2019-02-20:
Hello @MichaIng , Thank you for this very complete answer, 
Zerofree will work for me, I'll use it from a Linux machine.
I learned a lot about dietpi-backup, that will be useful to me, thanks !