MichaIng/DietPi #1088: Nvidia driver does not work properly on native X64

Invictaz, 2017-07-19:
After install it conflicts with Nouveau
I manually made an Xorg.conf with the Optimus settings but it cannot find the device
Also it's unable to launch the Nvidia driver

[Xorg.0.log.txt](https://github.com/Fourdee/DietPi/files/1159840/Xorg.0.log.txt)

I have tried everything described here

https://wiki.archlinux.org/index.php/NVIDIA_Optimus

Invictaz, 2017-07-19:
It seems there are some problems running on 4.9 kernel. Needs patches and GRUB patched.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=852152

https://devtalk.nvidia.com/default/topic/997549/kernel-4-10-1-1-and-340-102-driver-error-unable-to-load-the-kernel-module-nvidia-ko-/

Invictaz, 2017-07-19:
Might need 381.22

https://devtalk.nvidia.com/default/topic/1007999/unix-graphics-announcements-and-news/linux-solaris-and-freebsd-driver-381-22/

Invictaz, 2017-07-19:
However:

> On muxless Optimus laptops, or on laptops where a mux is present, but not set to drive the internal display from the NVIDIA GPU, the internal display is driven by the integrated GPU. On these systems, it's important that the X server not be configured to use the NVIDIA X driver after the driver is installed. Instead, the correct driver for the integrated GPU should be used. Often, this can be determined automatically by the X server, and no explicit configuration is required, especially on newer X server versions. If your X server autoselects the NVIDIA X driver after installation, you may need to explicitly select the driver for your integrated GPU.

http://us.download.nvidia.com/XFree86/Linux-x86/319.12/README/optimus.html

Invictaz, 2017-07-19:
Seems newer kernels than 4.? need a patch

https://devtalk.nvidia.com/default/topic/1007266/linux/patch-for-driver-375-39-with-stable-kernel-4-11/

Invictaz, 2017-07-20:
@Fourdee Might be the best solution to have Nouveau driver removed to avoid any conflicts. This might need adjustments to the ramdisk. Also, 381.22 is tried now via the unofficial .run installer. That requires GCC and MAKE and kernel-devel packages to work.

Invictaz, 2017-07-25:
@Fourdee It's alive 💃 

It now only works on HDMI out, not the internal laptop display. But that will come later.
Seems the newest 381.22 driver already makes a lot of effort to set up the right things. Although the i386 32 bit stuff isnt installed. So I might do that manually.

[image]

Invictaz, 2017-07-26:
@Fourdee It's alive 💃 

381.2 seems to do a lot more good than the 375.66 provided by Dietpi-install
It's for now only working via external HDMI but that will hopefully be fixed later.
Manually installing the Nvidia driver via the .run requires some additional packages. However it has to be done while X isn't running. And in dietpi-config Nvidia driver can be installed with X (and thus Nouveau) running. 

[image]

Invictaz, 2017-07-26:
384.59 is now out. Normally Nvidia isn't that fast with driver updates.
Might include a driver update in dietpi-update.

Fourdee, 2017-07-26:
@Invictaz 

>  It's alive 

Excellent 👍 

Yep, currently nvidia driver is apt-get install. I'll see if we can adjust to install latest via .run package.

Fourdee, 2018-04-05:
http://www.nvidia.co.uk/download/driverResults.aspx/133321/en-uk

> Note that many Linux distributions provide their own packages of the NVIDIA Linux Graphics Driver in the distribution's native package management format. This may interact better with the rest of your distribution's framework, and you may want to use this rather than NVIDIA's official package.

Unable to find a URL link, that links to the latest driver download. EG: we'd need to use a specific driver that would require manual updates to the link over time, that will always be behind the latest version:
http://www.nvidia.co.uk/content/DriverDownload-March2009/confirmation.php?url=/XFree86/Linux-x86_64/390.48/NVIDIA-Linux-x86_64-390.48.run&lang=uk&type=TITAN

We'll stick with APT driver, marking as closed.

closed 2018-04-05 by Fourdee