MichaIng/DietPi #1090: DietPi-Software | OpenBazaar locks up on language page

Fourdee, 2017-07-20:
> https://twitter.com/vp39216130/status/887943516202172417

Fourdee, 2017-07-21:
Tests:
- Fresh install of DietPi on RPi Zero W

Fourdee, 2017-07-22:
🈴 Issue with the OpenBazaar client, clicking next has no effect:
[image]

Fourdee, 2017-07-22:
No issues during our installation, everything compiled/installed fine. I believe this is a bug in the OB server and/or client.

MichaIng, 2018-09-04:
@Fourdee 
Not sure if it's related, but something I generally wanted to check throughout `dietpi-software` installs:
- `cp -R OpenBazaar-Server/* /etc/openbazaar-server/` does not copy hidden files/folders. For `.gitignore`, `.github/` and such, if no regular update via `git` is needed, this is no issue or even wanted, but OpenBazaar repo has at least one other hidden files that might affect the build/execution: https://github.com/OpenBazaar/OpenBazaar-Server/blob/master/.pylintrc
- Generally I suggest for `git` installs, that we clone them directly to destination folder and build them there (if needed). This has also the benefit, that on reinstalls, obsolete files are removed as well. The `.gitignore` skips config/data/build files well: https://github.com/OpenBazaar/OpenBazaar-Server/blob/master/.gitignore
Of course this would need to be checked for every software title where we switch to direct git clone to destination.

Fourdee, 2018-09-07:
@MichaIng 

Yep.

```git clone http://git.git /etc/wherewewantit``` should do it.

MichaIng, 2018-09-07:
@Fourdee 
Hmm just wanted to test on Stretch VM, but the service itself fails to start with OpenSSL error:
```
Sep 08 01:02:17 VM-Stretch systemd[1]: Started openbazaar.
Sep 08 01:02:18 VM-Stretch python[28180]: Traceback (most recent call last):
Sep 08 01:02:18 VM-Stretch python[28180]:   File "openbazaard.py", line 12, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from api.restapi import RestAPI
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/etc/openbazaar-server/api/restapi.py", line 5, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     import obelisk
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/obelisk/__init__.py", line 3, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from client import *
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/obelisk/client.py", line 3, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from zmqbase import ClientBase
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/obelisk/zmqbase.py", line 11, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet import defer, reactor
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/reactor.py", line 38, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet import default
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/default.py", line 56, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     install = _getInstallFunction(platform)
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/default.py", line 44, in _getInstallFunction
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet.epollreactor import install
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 24, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet import posixbase
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 18, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet import error, udp, tcp
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 29, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.internet._newtls import (
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/internet/_newtls.py", line 21, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from twisted.protocols.tls import TLSMemoryBIOFactory, TLSMemoryBIOProtocol
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/twisted/protocols/tls.py", line 41, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from OpenSSL.SSL import Error, ZeroReturnError, WantReadError
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/OpenSSL/__init__.py", line 8, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     from OpenSSL import rand, crypto, SSL
Sep 08 01:02:18 VM-Stretch python[28180]:   File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 124, in <module>
Sep 08 01:02:18 VM-Stretch python[28180]:     SSL_ST_INIT = _lib.SSL_ST_INIT
Sep 08 01:02:18 VM-Stretch python[28180]: AttributeError: 'module' object has no attribute 'SSL_ST_INIT'
Sep 08 01:02:18 VM-Stretch systemd[1]: openbazaar.service: Main process exited, code=exited, status=1/FAILURE
Sep 08 01:02:18 VM-Stretch systemd[1]: openbazaar.service: Unit entered failed state.
Sep 08 01:02:18 VM-Stretch systemd[1]: openbazaar.service: Failed with result 'exit-code'.
```
Can you replicate, perhaps something with libssl version?

Fourdee, 2018-09-08:
Interesting:
https://github.com/OpenBazaar/OpenBazaar-Server#note-this-project-is-no-longer-maintained-please-see-version-2-at-httpsgithubcomopenbazaaropenbazaar-go

Fourdee, 2018-09-08:
Ok installation and service now works, however, cannot find documentation on how to set the user/password for server.

Had no effect:
```
cat << _EOF_ > $G_FP_DIETPI_USERDATA/go/src/github.com/OpenBazaar/openbazaar-go/ob.cfg
USERNAME = openbazaar
PASSWORD = $GLOBAL_PW
_EOF_
```

Fourdee, 2018-09-08:
Online doc updated: https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1796#p1796

Everything seems to be running, however, unable to connect client.

Fourdee, 2018-09-08:
^^ low install count.

MichaIng, 2018-09-08:
@Fourdee 
> Interesting:
https://github.com/OpenBazaar/OpenBazaar-Server#note-this-project-is-no-longer-maintained-please-see-version-2-at-httpsgithubcomopenbazaaropenbazaar-go

Really good find 👍! Would have been a waste of time to fix a deprecated software version 🤣.

> unable to connect client.

Hmm, will try as well later.
_____
Tiny fix: https://github.com/Fourdee/DietPi/commit/d59cf68e629924c88740540f6fe357a6bb13d2dc

Fourdee, 2018-09-09:
@MichaIng 

> Btw: Starting/Restarting the service takes quite much time. Several compile and other shorter and longer init processes are starting, before everything calms down and the two final processes "go run" and "openbazaard start" stay in place. Takes as well 1-5 minutes depending on device, I guess, which should be added to our docs.

Yep, although, i didn't notice this with go 1.9 and ARMv7. Only go 1.11 and ARMv8.

Fourdee, 2018-11-04:
Still broken. Flagging for removal due to low install count.

MichaIng, 2018-11-05:
@Fourdee 
I am also doing final test currently. If without success I agree, but give me a change, since I still like the concept of this software.
Low install count is only expected, since it currently doesn't work 😉. Or is there a use without connected client?

MichaIng, 2018-11-05:
PR up: https://github.com/Fourdee/DietPi/pull/2224
- ToDo: Check Go v1.9
- Check possibility to hard code data dir, since otherwise this needs to be added to every openbazaard call.

NB: Automated user/pw addition is not possible, since related entries appear doubled and/or may appear at the and ob an json array or not, thus require a `,` sometimes, sometimes not. Having it, even with entry at end of array breaks OB start. I am not keen to rely on this default config always looking the same. And since user anyway manually needs to enter client IP, no harm to leave user/pass input via sage `go run openbazaard setapicreds` to user as well.
ToDo: `aSOFTWARE_REQUIRES_USERINPUT=1`

MichaIng, 2018-11-06:
@Fourdee 
PR ready. Tested on all distros (amd64), works well with new start method via compiled openbazaar-go binary within /mnt/dietpi_userdata/go/bin.
This starts much faster (especially on go 1.11) and allows easier end user access by adding this to $PATH as well.

Fourdee, 2018-11-08:
Test installs + UI:
- 🈯️ ARMv7 (ASUS TB)
- 🈯️ ARMv8 (C2)

> You call that a password? Try again.
Enter a veerrrry strong password:

🤣 

@MichaIng 

Works a charm, legend, great fix 👍

closed 2018-11-08 by Fourdee