MichaIng/DietPi #1091: DietPi-Software | Add "Crelay" installation option.

R1CH-T, 2017-07-21:
Please add a DietPi-Software option to install "Crelay": A Command-line / Webpage interface for various relay boards.

I will add to this soon, but have run out of time today.

R1CH-T, 2017-07-21:
From this morning's notes. Tested and working:

https://github.com/ondrej1024/crelay

```
Installation from source:

Dependencies: GIT, BUILD-ESSENTIAL.

apt install libftdi1 libftdi-dev libhidapi-libusb0 libhidapi-dev libusb-1.0-0 libusb-1.0-0-dev

REBOOT?

cd /mnt/dietpi_userdata

git clone http://github.com/ondrej1024/crelay

cd crelayhttps://github.com/ondrej1024/crelay

Installation from source:

Dependencies: GIT, BUILD-ESSENTIAL.

apt install libftdi1 libftdi-dev libhidapi-libusb0 libhidapi-dev libusb-1.0-0 libusb-1.0-0-dev

REBOOT?

cd /mnt/dietpi_userdata

git clone http://github.com/ondrej1024/crelay

cd crelay

cd src

make [DRV_CONRAD=n] [DRV_SAINSMART=n] [DRV_HIDAPI=n]

make install

REBOOT?

crelay

crelay -i

ADD crelay -D to system services

cd src

make [DRV_CONRAD=n] [DRV_SAINSMART=n] [DRV_HIDAPI=n]

make install

REBOOT?

crelay

crelay -i

ADD crelay -D to system services

```

Fourdee, 2017-07-21:
@R1CH-T 

Thanks for this Rich. I can get this added in to DietPi based on your notes, if you like? Or your more than welcome to try if you would like? Just let me know :)

> Failed and then worked after a reboot...

Shouldn't need a reboot, unless their are hardware changes (eg: ```config.txt```) etc.
Most likely the service needs restarting or checking for errors?

R1CH-T, 2017-07-22:
Hi.

I just rolled back DietPi on BerryBoot, re-ran the steps above and had no need to reboot. I think I must have missed out the `make install`, before testing it. It worked straight away.

I've just edited my above post, because it was a mess. These steps should work fine now.

I would write up the dietpi-software code for this, but it would be unnecessary at this point, IMO. It's pretty straightforward. 

What I am more interested in, at this point, are the steps involved in installing the system service and how we go about uninstalling, updating etc.

So, if you could go ahead and knock out the code for this program and post the steps, it will be very useful for me in getting an overview of the whole process of adding a feature to DietP. It might also prove very valuable to others as an example of how to add features, too. This is obviously not a simple apt-get install, like with say, `vim` or `mc`, but uses a `git clone` and `make install`, instead.

An improvement to this might be to be able to configure which relay card to control, via dietpi, by doing a `crelay -i` and using the output to create a system service for one of multiple relay boards.
(Unfortunately, at the moment, the daemon is only able to control one board and provide a webpage at `localhost:8000`)

For example, the little USB relay board that I have been using to test this shows up as so:
```
~$ sudo crelay -i

Detected relay cards:
  #1	HID API compatible relay card (serial BITFT)
```
Other detected cards will show up as consecutive entries in the list and could be selected by parsing the string after 'serial' and over/writing the service with the contents: `crelay -d -s <serial>`; in this case`crelay -d -s BITFT`.

For more information, just check out the project's [README.md](https://github.com/ondrej1024/crelay/blob/master/README.md).

Let me know what you think.
Rich :)

R1CH-T, 2017-07-22:
N.B.
I just tried running make uninstall, which didn't work, so i tried make -n install, which gave me:
```
root@DietPi:/mnt/dietpi_userdata/crelay/src# make -n install
echo "[Install binary]"
install -m 0755 -d		/usr/local/bin
install -m 0755 crelay	/usr/local/bin/crelay
```
I hope this helps.

R1CH-T, 2017-07-22:
One last thing, I promise:
I noticed [this note in the readme](https://github.com/ondrej1024/crelay#note-3-gpio-controlled-relays):

> Note 3 (GPIO controlled relays):
> 
> Since GPIO pin configuration is strictly device specific, the generic GPIO mode is disabled by default and can only be used in daemon mode. In order to enable it, the specific GPIO pins used as relay control lines have to be specified in the configuration file, [GPIO drv] section.

Might we be able, at some point, to also add a way to configure GPIO Relay Boards via the menu?
(Or is that best left up to the user; especially if the code changes, in the future?)
`/mnt/dietpi_userdata/crelay/conf/crelay.conf`:
```
################################################
#
# crelay config file
#
# This file is read by crelay in daemon mode
# from /etc/crelay.conf
#
################################################

# HTTP server parameters
################################################
[HTTP server]
server_iface = 0.0.0.0    # listen interface IP address
#server_iface = 127.0.0.1 # to listen on localhost only
server_port  = 8000       # listen port
relay1_label = Device 1   # label for relay 1
relay2_label = Device 2   # label for relay 2
relay3_label = Device 3   # label for relay 3
relay4_label = Device 4   # label for relay 4
relay5_label = Device 5   # label for relay 5
relay6_label = Device 6   # label for relay 6
relay7_label = Device 7   # label for relay 7
relay8_label = Device 8   # label for relay 8
pulse_duration = 1 	  # duration of a 'pulse' command in seconds
    
# GPIO driver parameters
################################################
[GPIO drv]
#num_relays = 8    # Number of GPIOs connected to relays (1 to 8)
#relay1_gpio_pin = 17   # GPIO pin for relay 1 (17 for RPi GPIO0)
#relay2_gpio_pin = 18   # GPIO pin for relay 2 (18 for RPi GPIO1)
#relay3_gpio_pin = 27   # GPIO pin for relay 3 (27 for RPi GPIO2)
#relay4_gpio_pin = 22   # GPIO pin for relay 4 (22 for RPi GPIO3)
#relay5_gpio_pin = 23   # GPIO pin for relay 5 (23 for RPi GPIO4)
#relay6_gpio_pin = 24   # GPIO pin for relay 6 (24 for RPi GPIO5)
#relay7_gpio_pin = 25   # GPIO pin for relay 7 (25 for RPi GPIO6)
#relay8_gpio_pin = 4    # GPIO pin for relay 8 ( 4 for RPi GPIO7)
    
# Sainsmart driver parameters
################################################
[Sainsmart drv]
num_relays = 4   # Number of relays on the Sainsmart card (4 or 8)
```

R1CH-T, 2017-07-27:
N.B.

The config file must be copied to `/etc/crelay.conf`

techdabbler, 2017-10-08:
@R1CH-T @Fourdee 

I took your above notes and got it working for install and as much deletion as your notes mention. I threw it up into a gist for the moment (trying to not make messes with loads of merges right now). 

https://gist.github.com/techdabbler/fe0fd4a78fac0bc8d141f6938d800272

I've tested it on x86_64 and it works. Need to test across ARM 6,7,8 to see if any of them don't work for starters. I made some changes most notably not using /mnt/dietpi_userdata to hold the source code before compiling. No need to I think? And a quick question esp for @Fourdee  -- as far as I know I have to cd into the directory to make / make install.  I added a cd $HOME to return us back re:comment from another commit is there a better way of doing this?

I'll need some help on what is best to put into any statically loaded crelay.conf.

If you download it to test (PLEASE DO!) make sure you nano /DietPi/dietpi/dietpi-services and add 'crelay' to the list -- or it will not automatically start on reboot.

Let me know your thoughts. Oh and the index is 163 if you want to fast search for it review anything.

R1CH-T, 2017-10-10:
Thanks for taking a look at this @techdabbler.

I've just checked out your gist and it looks good. I will try it out when I get around to it; hopefully soon.

Please check out https://github.com/ondrej1024/crelay/issues/11 as I have asked @ondrej1024 to fix a couple of issues in Crelay, over there. I posted a demonstration there, of Crelay working with an 8-relay GPIO card, too.

> I'll need some help on what is best to put into any statically loaded crelay.conf.

Just let me know what you need. :)

There is an example crelay.conf attached to bottom of the post I mentioned, above: It is configured for the 8x relay GPIO board.

techdabbler, 2017-10-10:
@R1CH-T Thanks. Looking further I see the crelay project actually provides the init.d and the conf file in the repo -- I didn't realize that.  I'll flip it over to those supported files when I get time (away from systemd script). It turns out that with systemd setup crelay -D appears running but I can't reach it via the web interface but manually it would work.

RE Conf file the real question is what do we want the defaults to be? Since it appears the conf file has to list each one.  Also, we have to uncomment each line correct to make it work?  Or just say hey folks -- go edit once installed.

R1CH-T, 2017-10-14:
> It turns out that with systemd setup crelay -D appears running but I can't reach it via the web interface but manually it would work.

It's odd that you cannot reach the web interface: Maybe a network issue at your end?
It works fine for me at the moment.

> RE Conf file the real question is what do we want the defaults to be? Since it appears the conf file has to list each one. Also, we have to uncomment each line correct to make it work?

(TL;DR)
The top section of the config file (under [HTTP server]) is used to provide network settings and custom names for relays (i.e. the name of the device that you are turning on and off) and don't need altering unless you wish to do so. When a USB relay board is plugged in, Crelay will autodetect the number of relays present and only show the triggers for those that are present in the web interface.

The commented out lines, below, under [GPIO drv] are only useful if you are using a GPIO relay board. As there is currently no way for Crelay to detect relays connected in this way, these must be manually set for each GPIO connected relay. giving the correct pin number for each. You must also state how many relays are connected, in order to show the right number of switches in the web interface.

> Or just say hey folks -- go edit once installed.

At the moment, that's how it stands, yes. However, It should be a relatively simple matter to create a configuration option in DietPi-Config, with menu options to alter this file. What do you think?

> It also looks like your requests for issue fixes shouldn't impact the dietpi installation/uninstall sections once it is merged into crelay's main branch but might alter config? Is that an accurate reading?

Several files have been altered in order to fix the outstanding issues and have yet to be merged into the main branch of Crelay. My request to add support for boards larger than 8x relays will also require a fair few changes, but I'd imagine that a simple uninstall then reinstall of Crelay would bring it up-to-date.

R1CH-T, 2017-10-14:
> ...what do we want the defaults to be?

I still have the test set-up that I demonstrated in the [Crelay issue page](https://github.com/ondrej1024/crelay/issues/11), so I just tried something out:

With all eight relays still connected to the GPIO ports, I ran crelay -D and accessed the web interface. All eight relays showed up and functioned correctly. I then plugged in my 2x relay USB card and refreshed the page and the interface immediately switched to showing two switches, which functioned to toggle the relays on this newly introduced card. This means that USB cards take priority and will override the GPIO config when they are detected.

This means, conceivably, that if the default was to have everything uncommented initially; any combination of relays would supported "Out of the box".

I would be extremely wary of doing this, though, as anything else plugged into the GPIO header would be at risk of being damaged.

techdabbler, 2017-10-14:
@R1CH-T 

> It's odd that you cannot reach the web interface: Maybe a network issue at your end?
It works fine for me at the moment.

Interesting.  It works for me if I run crelay -D manually but if I allow it to autostart I don't see it.  Are you using autostart (ie - systemd) and what ARCH are you on? I'm testing on x86_64 I wonder if it works on ARM?

> When a USB relay board is plugged in, Crelay will autodetect the number of relays present and only show the triggers for those that are present in the web interface.
Great!

> Several files have been altered in order to fix the outstanding issues and have yet to be merged into the main branch of Crelay. My request to add support for boards larger than 8x relays will also [...]

Are you comfortable with sending this into dietpi now with the above info or would you think it is better than hang tight until the fixes go into place? It sounds like the larger board is a feature upgrade but the high/low pin thing is more of a bug fix?  Thoughts?

> This means, conceivably, that if the default was to have everything uncommented initially; any combination of relays would be supported "Out of the box".

> I would be extremely wary of doing this, though, as anything else plugged into the GPIO header would be at risk of being damaged.

Hmm. On one hand I like the idea of configuration-less. Is this dangerous because it might detect non-relays as relays and allow control of pins that shouldn't be allowed? I'm not quite gathering the danger I admit.

R1CH-T, 2017-10-22:
@techdabbler 

Sorry for taking a week to reply, but I simply forgot about it. I read your post last weekend, sat down to formulate a response and quickly fell asleep. I just re-opened and refreshed this tab in my browser and was astonished to realise that I hadn't actually responded to you.

I hope I can remedy that mistake. So...

> Interesting. It works for me if I run crelay -D manually but if I allow it to autostart I don't see it. Are you using autostart (ie - systemd) and what ARCH are you on?

I wish I'd spotted this sooner; but yes: I have been running `crelay -D` or `-d` manually, too.

> I'm testing on x86_64 I wonder if it works on ARM?

Yes. It works fine on several devices: I have tested it on a desktop, a laptop (x86_64) and a RasPi 0,1,2 and 3 (ARM).

> Are you comfortable with sending this into dietpi now with the above info or would you think it is better than hang tight until the fixes go into place?

I have no direct input in regards of what ends up in DietPi. I'm just an enthusiast/contributor/donor to this project. Dan (@Fourdee) makes all the decisions in this regard, but he is currently in the process of dealing with [something akin to an alien invasion](https://github.com/Fourdee/DietPi/labels/ARMbaby) at the moment, so this issue is currently not a top priority. Hang tight. ;)

>  It sounds like the larger board is a feature upgrade but the high/low pin thing is more of a bug fix? Thoughts?

Yes and Yes.

There's not much that I can do about that one either. The project leaders and developers do what they can, but Life has an annoying habit of getting in the way. :grrrr:

> Hmm. On one hand I like the idea of configuration-less. Is this dangerous because it might detect non-relays as relays and allow control of pins that shouldn't be allowed? I'm not quite gathering the danger I admit.

It's not going to detect anything. If you have other things plugged into other GPIO pins, it's probably not a good idea for Crelay to be 'owning' those same pins. When using GPIO, there's no microprocessor in-between the Raspberry Pi (or whatever) and the relay board. Imagine clicking on a button in the the web interface and sending +5v to an expensive and very sensitive sensor device that you have plugged into that pin. It's not the end of the world, but these tend to be pretty dumb devices that can be broken very easily. If you don't know what you're doing and start introducing electrical current through the wrong places, you will kill your hardware. That's the danger. :$$$:

I'm sorry for the delay and hope this helps.
Rich. :)

techdabbler, 2017-10-23:
@R1CH-T Thanks for the reply. No worries on the delay at all.

Good to know that you are starting it manually in that means the systemd file I created isn't working. Oh well. I think I found the init file you original posted on the github repository so I can grab that out of the clone and use that instead of systemd. I imagine it works. So that should fix that problem.

As far as my question about sending it into dietpi now I was more referring to if in your opinion the bugs you submitted to the crelay team are serious enough that we should wait to introduce it as an 'optimized installation' in dietpi? I haven't really used it enough to make an informed decision in that regard if left to me I'd submit it as is once it gets installed and is running in a daemon mode.  I suppose I'm curious if you think otherwise?

RE: config -- I think I'm starting to see how this all connects.  There are 2 types of boards crelay supports -- USB and GPIO.  This is the part I believe I missed that got me all confused I couldn't figure out the relationship of GPIO pins to USB.  They are separate! Ah ha! So if we just include the http info in the configuration then  USB boards will be recognized and the appropriate number of relays will be displayed on the web panel? (Is this part correct?)  

While the GPIO section of the config supports GPIO boards -- ie not usb! Ah ok.  I see what you were saying now about uncommenting it all would support all out of the box. And yes I agree it is bad bad idea to make the GPIO active in that way without user intervention.  

If the above is true then I certainly think we should start with just the http section and everything else uncommented.

R1CH-T, 2017-10-26:
> As far as my question about sending it into dietpi now I was more referring to if in your opinion the bugs you submitted to the crelay team are serious enough that we should wait to introduce it as [...]

There's no massive hurry to add this at the moment. Until the ["GPIO Fix" version](https://github.com/ondrej1024/crelay/tree/gpio_fix) is merged into the main branch of Crelay, it might be worth holding back a little while.

> So if we just include the http info in the configuration then USB boards will be recognized and the appropriate number of relays will be displayed on the web panel? (Is this part correct?)

USB boards are detected and work "out of the box", so to speak. Any further modification is optional.

> If the above is true then I certainly think we should start with just the http section and everything else uncommented.

This seems wise at the moment.

R1CH-T, 2017-11-08:
Just so you guys know, @ondrej1024 [has merged]([url](https://github.com/ondrej1024/crelay/issues/11#issuecomment-341722589)) the "GPIO Fix" branch into the master repository, so everything is working fine now:

> @R1CH-T, I just released this fix to the master branch.
As for the patch allowing 16 GPIO relays, I need some more time for testing.

For now, Crelay should 'be good to go', so @Fourdee; if you have the time, the installation method worked out above should be adequate to get this thing working.

MichaIng, 2018-12-25:
Added to FeatHub: https://feathub.com/Fourdee/DietPi/+3

MichaIng, 2021-09-24:
We decided to cleanup open issues a bit and close those which will obviously not addressed soon. The FeatHub request stays however open to allow votes and if anyone wants to start working on implementing this software, please feel free to to a start, and we'll help where we can.

Here you find instructions about how to implement a new software title to `dietpi-software`: https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

closed 2021-09-24 by MichaIng