MichaIng/DietPi #1094: Add commit version number when using testing branch

Invictaz, 2017-07-23:
Since I'm running v154 from testing branch and v154 final is now released it would be nice if updating from testing branch shows the exact build instead of v154. Because v154 is in my case a test build and not the final.

Or make it v154beta or v154rc or v154-332a2

Something like that

Fourdee, 2017-07-23:
@Invictaz 
Aside from testing purposes, the testing branch should not be used and is potentially unstable. We cant support an in-devolpment branch across version releases. 
Code is constantly changing, patches are added/modified that your currently installation will miss etc.

Ideally, following process to create a restore point of master, then allow for testing branch tests:
- Install fresh image, keeping master branch
- backup system ```dietpi-backup 1```
- change ```/DietPi/dietpi.txt``` to ```gitbranch=testing```
- ```dietpi-update```
- Now testing can begin. 

Once finished (or the testing branch has more changes), you can:
- restore system ```dietpi-backup -1; reboot```
- Switch to testing branch and test again.

> Or make it v154beta or v154rc or v154-332a2

Version number is integer based currently. Once I update the testing branch to v155, you'll see the available update.

Invictaz, 2017-07-23:
Testing is never finished :P

MichaIng, 2018-01-31:
Close as question was answered, reopen on demand.

closed 2018-01-31 by MichaIng

Fourdee, 2018-01-31:
@MichaIng 

On fire with the spring cleaning, great work, really appreciate it 👍