MichaIng/DietPi #1096: Can't install Pihole on fresh install

FutureTense, 2017-07-23:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?154
- Ordoid C2
- USB Power
- EEMC
- Distro (EG: Jessie) | ```uname -a```?Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title: Pihole
- Can this issue be replicated on a fresh installation of DietPi?: Yes

#### Expected behaviour: App should be installed

Getting errors on install.  Tried manual install of Pihole using `curl -sSL https://install.pi-hole.net | bash
which gave an error and suggested ` Running `apt-get update`.  shows that the chace is not being updated.

#### Steps to reproduce:
Fresh install on Odroid C2.  Go through updates.  Set static IP (or do it during Pihole install, makes no difference).  Install Pihole.

OR

```
root@DietPi:~# apt-get update
Hit http://security.debian.org jessie/updates InRelease
Get:1 http://www.deb-multimedia.org jessie InRelease [35.1 kB]
Hit http://fuzon.co.uk all InRelease
Ign http://ftp.debian.org jessie InRelease
Ign http://www.deb-multimedia.org jessie InRelease
Hit http://fuzon.co.uk jessie InRelease
Hit http://ftp.debian.org jessie-updates InRelease
Hit http://security.debian.org jessie/updates/main arm64 Packages
Hit http://fuzon.co.uk all/c2 arm64 Packages
Hit http://ftp.debian.org jessie-backports InRelease
Hit http://security.debian.org jessie/updates/contrib arm64 Packages
Hit http://fuzon.co.uk all/main arm64 Packages
Hit http://security.debian.org jessie/updates/non-free arm64 Packages
Hit http://ftp.debian.org jessie Release.gpg
Hit http://security.debian.org jessie/updates/contrib Translation-en
Hit http://fuzon.co.uk all/testing arm64 Packages
Get:2 http://ftp.debian.org jessie-updates/main arm64 Packages/DiffIndex [8,392 B]
Get:3 http://www.deb-multimedia.org jessie/main Translation-en [39.2 kB]
Hit http://security.debian.org jessie/updates/main Translation-en
Hit http://ftp.debian.org jessie-updates/contrib arm64 Packages
Hit http://security.debian.org jessie/updates/non-free Translation-en
Get:4 http://www.deb-multimedia.org jessie/non-free Translation-en [2,933 B]
Get:5 http://ftp.debian.org jessie-updates/non-free arm64 Packages/DiffIndex [736 B]
Hit http://ftp.debian.org jessie-updates/contrib Translation-en
Get:6 http://ftp.debian.org jessie-updates/main Translation-en/DiffIndex [3,196 B]
Get:7 http://ftp.debian.org jessie-updates/non-free Translation-en/DiffIndex [736 B]
Get:8 http://ftp.debian.org jessie-backports/main arm64 Packages/DiffIndex [27.8 kB]
Hit http://fuzon.co.uk jessie/backports arm64 Packages
Get:9 http://ftp.debian.org jessie-backports/contrib arm64 Packages/DiffIndex [22.4 kB]
Hit http://fuzon.co.uk jessie/main arm64 Packages
Get:10 http://ftp.debian.org jessie-backports/non-free arm64 Packages/DiffIndex [8,530 B]
Get:11 http://ftp.debian.org jessie-backports/contrib Translation-en/DiffIndex [7,468 B]
Get:12 http://ftp.debian.org jessie-backports/main Translation-en/DiffIndex [27.8 kB]
Get:13 http://ftp.debian.org jessie-backports/non-free Translation-en/DiffIndex [16.7 kB]
Hit http://ftp.debian.org jessie Release
Err http://www.deb-multimedia.org jessie/main arm64 Packages
  404  Not Found
Err http://www.deb-multimedia.org jessie/non-free arm64 Packages
  404  Not Found
Hit http://ftp.debian.org jessie/main arm64 Packages
Ign http://www.deb-multimedia.org jessie/main Translation-en_GB
Hit http://ftp.debian.org jessie/contrib arm64 Packages
Ign http://www.deb-multimedia.org jessie/non-free Translation-en_GB
Hit http://ftp.debian.org jessie/non-free arm64 Packages
Hit http://ftp.debian.org jessie/contrib Translation-en
Hit http://ftp.debian.org jessie/main Translation-en
Hit http://ftp.debian.org jessie/non-free Translation-en
Ign http://fuzon.co.uk all/c2 Translation-en_GB
Ign http://fuzon.co.uk all/c2 Translation-en
Ign http://fuzon.co.uk all/main Translation-en_GB
Ign http://fuzon.co.uk all/main Translation-en
Ign http://fuzon.co.uk all/testing Translation-en_GB
Ign http://fuzon.co.uk all/testing Translation-en
Ign http://fuzon.co.uk jessie/backports Translation-en_GB
Ign http://fuzon.co.uk jessie/backports Translation-en
Ign http://fuzon.co.uk jessie/main Translation-en_GB
Ign http://fuzon.co.uk jessie/main Translation-en
Fetched 201 kB in 6s (32.0 kB/s)
W: GPG error: http://www.deb-multimedia.org jessie InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 5C808C2B65558117
W: Failed to fetch http://www.deb-multimedia.org/dists/jessie/main/binary-arm64/Packages  404  Not Found

W: Failed to fetch http://www.deb-multimedia.org/dists/jessie/non-free/binary-arm64/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
root@DietPi:~#
```

#### Did you submit a ```dietpi-bugreport```?

Ref code  5d35b65f-dab9-4588-b41e-279b137a7a3f-0  

#### Extra details:

FutureTense, 2017-07-23:
Just for kick, I installed the original OS that came with the odroid (fortunately I saved a copy of the image) and pihole installed, so it's not an issue with my network.

`Linux odroid64 3.14.79-108 #1 SMP PREEMPT Mon Feb 27 23:18:26 BRT 2017 aarch64 aarch64 aarch64 GNU/Linux`

Fourdee, 2017-07-24:
Hi,

Thanks for the report 👍 , we'll take a look.

Appears the deb multimedia repo is causing apt to fail, should be a simple fix, but i'll let you know.

Fourdee, 2017-07-24:
Yep, appears to be an issue with the debian multimedia repo, arm64 packages are no longer available:
http://www.deb-multimedia.org/dists/jessie/main/binary-arm64/Packages

[image]

## solution:
Remove the multimedia repo, not required for DietPi:
```
rm /etc/apt/sources.list.d/deb-multimedia.list
apt-get update
```

Fourdee, 2017-07-26:
We'll remove the apt mirror during v155 patch.

Fourdee, 2017-07-26:
Completed.

closed 2017-07-26 by Fourdee