MichaIng/DietPi #1097: SickRage wrong repo

donbrew1, 2017-07-24:
I attempted to install SickRage via dietpi 153. 

 Evidently dietpi is using the git that has been taken down due to DMCA issues.  The one that works is : https://git.sickrage.ca/SiCKRAGE/sickrage.git  WEBSITE: sickrage.ca

Fourdee, 2017-07-24:
@donbrew1 

Thanks for the report 👍 

I'am well aware of the DMCA takedown notice. Sickrage are disputing this, we'll await the outcome:
https://www.reddit.com/r/usenet/comments/6oc65t/sickrage_github_repo_dmcad/

The owner of the repo you listed is "echel0n", who is claimed to be the DMCA originator.

donbrew1, 2017-07-24:
FWIW, I like echel0n's version better.

DDA78, 2017-11-11:
Can I change the repo myself, any help will be appreciated, I am looking to try pymedusa instead of sickrage, it's much more active, sickrage seems dead

Fourdee, 2018-02-22:
We'll stick with https://github.com/SickRage/SickRage.

Marking as closed.

closed 2018-02-22 by Fourdee