MichaIng/DietPi #1098: PiBakery configuration compatibility

Invictaz, 2017-07-27:
Maybe it's handy that people who don't have a mouse / keyboard can setup DietPi before first launch. PiBakery is intended for this first setup but lacks some of the building blocks that Dietpi-config has.  

https://github.com/davidferguson/pibakery

Fourdee, 2017-07-27:
@Invictaz 

> Maybe it's handy that people who don't have a mouse / keyboard can setup DietPi before first launch. 

We also have a automation system to achieve this:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=273#p1140

Invictaz, 2017-07-28:
@Fourdee not correct as the /boot/dietpi.txt is IN the image and thus on a EXT4 partition you cannot write on from Windows. Correct me if I'm wrong.

sytone, 2017-07-28:
I use ext2fsd to do any changes like this http://www.ext2fsd.com/

closed 2018-02-22 by Fourdee