MichaIng/DietPi #1099: Mono release not compatible with optimized software release (Emby)

DncrJim, 2017-07-29:
## Maybe not bug so much as Installation Compatibility Disagreement? 

#### Required Information:
- DietPi Version | V154
- SBC Device | RPi 3

#### What I am doing:
Installing emby-server through Software-Optimized on fresh diet-pi install.
Had same issue with V153 (and all versions of emby-server 3.2.23-3.2.26)

#### What's Happening:
emby-server crashes upon media scan with an 'out of memory' error. Suggested/Expected problem as learned from the emby forum is incompatible version of mono. 

#### Possible solution?:
Emby release is only compatible up to mono 4.8.1, but dietpi installs 5.x (and I'm not knowledgeable enough to be able to roll it back, apparently) *looks forlornly at poor pi that has endured 3 fresh installs today that I have promptly screwed up beyond repair*

Is there any way either diet-pi can be aware of this information and correct for it upon install, and/or have a prompt to ask? alternatively, instructions or a link in the right direction for me to downgrade mono after a fresh install without breaking all of the dependencies would be amazing. Google has let me down something fierce today...

Thank you as always.
~Jim

Fourdee, 2017-07-29:
@DncrJim 

Hi Jim,

Thanks for the report 👍 

Do you have any other installed software on the device, that may be using up additional memory?
```
dietpi-services status
```

I personally dont believe downgrading Mono versions is the solution here. Generally speaking "newer" versions of libraries provide more stability and improvements.

Looking into increasing the swapfile size may prove a fix. Lets check the current size with:
```
/DietPi/dietpi/func/dietpi-set_dphys-swapfile
```

---

> emby-server crashes upon media scan with an 'out of memory' error.

Roughly how many media files are contained within the directory?

DncrJim, 2017-07-29:
I agree that having the old version of mono is unfortunate. 
only other running service is samba, current swap is 54, which now that I'm reading that seems really stupid-small, especially with mono libraries? I haven't changed the swap, that's what it gave me on the fresh V154 install. I am sure I can figure out how to increase the swap, but I can't seem to find any recommendations on swap size. Are there any insights out there for this application?

Scanning somewhere in the neighborhood of 1700 files in the first directory, and another 12k or so in a second. (I'm only getting 100 or so files in on a scan) All files are on a smb share on the network, not on the device. 

I am afraid I may have jumped the gun with this report. While rolling back mono did reduce emby-server crashing, and others have historically fixed a number of problems by rolling back mono upon the advice of the devs at Emby, my underlying issue with the scan does not seem to have been solved. Sometimes the scan itself will lock up but emby will continue to run, sometimes it is still crashing emby.

As mentioned, this has solved some problems for others in the past, so the option might want to be considered. It also definitely assisted in the catastrophic failure, but it appears there are more layers to this onion at least for me.

Fourdee, 2018-04-05:
As mono is a shared library, installed for many programs in ```dietpi-software```, changing the version for one specific item, may break other software installations that use mono.

As such, we'll stick with the mono version as provided by its repo.

I'll mark this as closed.

closed 2018-04-05 by Fourdee