MichaIng/DietPi #1100: Information: DietPi development on temporary hold (paternity leave)

Fourdee, 2017-07-29:
Hi all,

Just a heads up, development on DietPi has been put on hold, and my activity will be reduced for a few weeks. Expecting the baby 👶 to arrive in the next day or two.

I'll monitor tickets when possible, and, keep the ship running in the mean time, however, should I not respond, everyone knows why 😄

dhernan3, 2017-08-02:
My congratulations!!!
My best wishes for all your family and new baby!

Fourdee, 2017-08-04:
[image]

dhernan3, 2017-08-04:
Congratulations!

elRadix, 2017-08-04:
@Fourdee is a Happy Father now, congratulations with your new born!!

Lanxon, 2017-08-04:
Congratulations...best day ever for you..

Invictaz, 2017-08-06:
Congrats

WolfganP, 2017-08-06:
Congrats @Fourdee. So I assume this is the newer release, Fivedee? :-)

Fourdee, 2017-08-06:
@WolfganP 

Thanks 👍 

> Fivedee? :-)

Lol, second baby. 6D ;)

k-plan, 2017-08-11:
@Fourdee 

Congratulations and regards to mother, father, son to the new member of the family.

btw, nice name.

Elluminatus, 2017-08-17:
Congrats and all the best for your family...

alexthe-red, 2017-08-20:
congrats on the new branch!