MichaIng/DietPi #1101: Tonido doesn't install (and FIX)

dhernan3, 2017-08-02:
Tonido doesnt' install, failed to download libjpeg8. Tried on Raspberry Pi3 and VM.

Unable to Install Tonido: lightweight cloud based backup system   http://ftp.us.debian.org/debian/pool/main/libj/libjpeg8/libjpeg8_8d1-2_amd64.deb is offline and/or unreachable. 

The problem is the package trying to install doesn't exist. The package must be called:
ibjpeg8_8d-1+deb7u1_amd64.deb      for VM
libjpeg8_8d-1+deb7u1_armhf.deb       for RPi 3

Fourdee, 2017-08-06:
@dhernan3 

Thanks for the report 👍 

Most likely the required package version has changed, no longer available. We'll take a look.

Fourdee, 2017-08-06:
@dhernan3 

This should of been resolved in v154: https://github.com/Fourdee/DietPi/issues/1059

Is the issue still occurring with v154 installed?

Fourdee, 2017-08-06:
@dhernan3 

Fixed for v155, can be patched now with:
```
wget https://github.com/Fourdee/DietPi/raw/testing/dietpi/dietpi-software -O /DietPi/dietpi/dietpi-software
```
Tonido will now install.

closed 2017-08-06 by Fourdee