MichaIng/DietPi #1102: DietPi version for Rock64

douglasmiranda, 2017-08-05:
So, any plans for support / make an DietPi for [Rock64](https://www.pine64.org/?page_id=7147)?

[Here](http://wiki.pine64.org/index.php/ROCK64_Main_Page) we can see it's already running Debian Strech Linux Kernel 4.4.

PS: I'm asking here, because I'm waiting for a long time the approval of my account on DietPi forum, so far nothing.

Fourdee, 2017-08-06:
@douglasmiranda 

Ordered one recently, its in transit:
[image]

When it arrives, we'll support the board and provide an image.

k-plan, 2017-08-11:
Hmm, seem to have some nice hardware features, which RPi users have to wait for a long time ...

But Rockchip RK3328 Quad-Core ARM Cortex A53 64-Bit Processor ?

Hope, it will be not the same non-starter like PINE A64 for the community ...

Fourdee, 2017-08-18:
notes:
- 🈴 https://github.com/ayufan-rock64/linux-build/releases/download/0.4.17/stretch-minimal-rock64-0.4.17-85-arm64.img.xz hangs on boot ```[7.x]``` for ```eth0 device MAC address```

alexthe-red, 2017-08-20:
+1 for rock64. DietPi has ruined me f0r all non-compatible boards out there :)

Fourdee, 2017-08-26:
#### Image available for beta testing:
#### http://dietpi.com/downloads/testing/DietPi_Rock64-arm64-(Stretch).7z

#### Known issues:
#### https://github.com/Fourdee/DietPi/issues/1102#issuecomment-323343966

molemantis, 2017-08-28:
Thanks for this beta! As commented in your notes, SSH hangs when launching dietpi-software (works when directly connected to console) using either ssh or dropbear, and I wasn't able to boot after flashing to EMMC. Everything else is working great.

DeviantEng, 2017-09-25:
So glad to see there's work on this already.  Like others, for me it's DietPi or bust when it comes to Linux on SBC's.  Definitely hope eMMC booting is supported as well.  I received a ROCK64 today, but I believe it's DoA so I've still got a couple of weeks before I can contribute any testing.

alexthe-red, 2017-09-25:
Something that threw me off for a bit with this board and Dietpi:: 

- Couldn't access via SSH unless fresh dietpi install had ethernet settings set to 10. Pine forums mention that you can physically accomplish this by snipping a couple wire pairs on your ethernet cable. Once inside, I was bumped the ethernet speed to 100 with `ethtool -s eth0 speed 100 duplex full`. Too afraid to bump it to 1000 since the Pine64 board couldn't handle it (http://dietpi.com/phpbb/viewtopic.php?f=11&t=833)

- The memory card I was using (32gig) wouldn't correctly partition to use all the space, so consequently it would fill up during  the auto updates on first boot. I fixed this using the following thread:
https://www.raspberrypi.org/forums/viewtopic.php?t=45265

Should I just be posting these as issues? Or are they separate because of the testing status?

Fourdee, 2017-10-13:
Image updated for Official Release:
- http://dietpi.com/downloads/images/DietPi_Rock64-arm64-(Stretch).7z

Includes latest kernel and fixes networking issues. Also includes working ayufan Rock64 repo (for kernel updates).

Existing users should reinstall with this image, as patching is not possible.

Fourdee, 2017-10-13:
Performance notes:
- SD:
[image]
- LAN
```
root@DietPi:~# iperf -c 192.168.0.111 -p 5001 -t 30 -d
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
------------------------------------------------------------
Client connecting to 192.168.0.111, TCP port 5001
TCP window size:  425 KByte (default)
------------------------------------------------------------
[  3] local 192.168.0.20 port 57508 connected with 192.168.0.111 port 5001
[  5] local 192.168.0.20 port 5001 connected with 192.168.0.111 port 55342
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-30.0 sec  1.15 GBytes   330 Mbits/sec
[  5]  0.0-30.0 sec  5.29 GBytes  1.51 Gbits/sec
```

DeviantEng, 2017-10-13:
Will be testing eMMC with this board early next week.

BryanSmithDev, 2017-10-17:
Is 3.5mm audio a known issue? I get nothing but noise from it. However I am running video out HDMI so might that be the issue? HDMI claiming the audio over 3.5mm? I have selected 3.5mm in diet-pi config. I am on v156 however. The index issues that are fixed via v157 might have fixed this.

Fourdee, 2017-10-29:
@BryanSmithDev 

> EDIT: Tried v157. Still no audio.

Thanks 👍  looks like the audio cards/devices have changed since last kernel.
```
hw:0,0 = HDMI
hw:1,0 = 3.5mm
```

Fourdee, 2017-10-29:
[image]

Fourdee, 2017-10-29:
Live on http://dietpi.com , with known outstanding issue with audio which will be patched in v158: https://github.com/Fourdee/DietPi/issues/1102#issuecomment-340279988

closed 2017-10-29 by Fourdee

douglasmiranda, 2017-11-02:
I'm trying to run on my Rock64, but everytime dietpi-update runs I get the error: 

insufficient space to run rootfs

molemantis, 2017-11-02:
You need to resize your root partition to use all available disk space. 

Here are the basic commands I used: 

gdisk /dev/mmcblk1
w 
Y

fdisk /dev/mmcblk1
d
choose / partition

n
choose / partition 
w

partprobe
resize2fs /dev/mmcblk1p7 (7 being / )

douglasmiranda, 2017-11-03:
Great! Thanks xD

xnathanh, 2017-12-22:
@molemantis @douglasmiranda i got error thing 

> insufficient space to run rootfs

>(I'm using Etcher to flash the sd card)
 but i try to resize root partition,doesn't work,could you send more detail of resize root partition script?

Fourdee, 2017-12-23:
Reference:
https://github.com/Fourdee/DietPi/issues/1333#issuecomment-353661049

darthyoh, 2018-01-14:
Hi !
I'm interested in getting a rock 64 (powered by dietpi)
First, thanks for the job done with it.
I just have a question : one of the reasons i'm interested with it is the emmc module to increase transfert rate. I imagine installing dietpi on a emmc module. Do you think this process http://dietpi.com/phpbb/viewtopic.php?f=8&t=980 can be applied to the rock 64 board ?

joshuamallow, 2018-01-25:
The latest v156 image hangs on first boot. Anyone else experiencing this? SD card works fine with other Debian images, but can't get DietPi to boot. Any thoughts?

Fourdee, 2018-01-25:
@joshuamallow 

Hi,

Please try the v6.0 RC image:
http://dietpi.com/downloads/testing/v6.0/DietPi_Rock64-ARMv8-Stretch.7z

joshuamallow, 2018-01-27:
Can you update the v6.0 RC image? The compression is corrupt.

Fourdee, 2018-01-27:
@joshuamallow 

> Can you update the v6.0 RC image? The compression is corrupt.

Confirmed, apologies, thanks for letting us know 👍 

Image has been re-compressed and uploaded:
http://dietpi.com/downloads/testing/v6.0/DietPi_Rock64-ARMv8-Stretch.7z

[image]

joshuamallow, 2018-01-28:
Thanks!

Working great...no issues so far. Is this going to be the final version once v6 goes to production?

Fourdee, 2018-01-28:
@joshuamallow 

Yep. There will be a minor update today, to bring it up to v6.0 release.

ViRb3, 2018-08-26:
@Fourdee the latest image fails to boot on a Rock64 1GB board. I believe this may be related to: https://github.com/armbian/build/commit/475f0cbd8700db23d882faca891f9c7196e824c0
I never managed to boot Android or Armbian either.