MichaIng/DietPi #1105: Dietpi hangs at boot after first-run upgrade | XU4

joaofl, 2017-08-09:
After flashing the image DietPi_v151_OdroidXU4-armv7-(Jessie), during first time boot, the Dietpi updates itself from 151 to 154 and reboots. Next, it asks for softwares to be installed, and without selecting any, I click Go, what leads to an "apt upgrade".

The first upgrade takes some 20 min, and after it, it reboots automatically, but hangs during boot on the following screen:
[image]

If I unplug it, and plug again, it does not boot anymore.

Happened 3 times in a row, with different SD cards, and with/without external USB drives plugged into the USB.

** the same does not happen with the image containing the kernel 4.9

Fourdee, 2017-08-10:
@joaofl 

Thanks for the report 👍  I'll take a look.

Fourdee, 2017-08-10:
No drives/partitions listed in ```blkid```

Fourdee, 2017-08-10:
@joaofl 

Image updated: https://github.com/Fourdee/DietPi/issues/1105#issuecomment-321542109

Thanks for reporting this 👍

Fourdee, 2017-08-10:
Completed, however existing installations will need to manually update their ```boot.ini``` after kernel upgrade:
https://github.com/Fourdee/DietPi/issues/1105#issuecomment-321542109

closed 2017-08-10 by Fourdee

kevinhaendel, 2017-08-28:
### I tried the following:
First, I used this image here: http://dietpi.com/downloads/images/DietPi_OdroidXU4-armv7-(Jessie).7z , and then I replaced the boot.ini file **before** and **after** "apt-get upgrade". But in both cases it is still not working.

### I used a workarount until this is working again:
I typed "apt-mark hold uboot" to prevent uboot from upgrading, this solves my problem, but I hope I can update it again in the future.

joaofl, 2017-08-28:
At some point the new image provided by @Fourdee worked out. But yesterday it seems that upgrading the uboot caused some issue. Again it hanged during boot. Not even the hearbeat LED blinks.

Not sure what to do.