MichaIng/DietPi #1106: SABnzbd UNRAR version 0.00

k3anton, 2017-08-09:
## Creating a bug report/issue:

Hi,

This has been mentioned in issue #865 but remains a problem on a fresh install.

#### Required Information:
- DietPi Version 154
- RPi 2, model b, armv71.
- RPi Power supply (original)
- MicroSD HC, Class 4, 8GB / 1TB external HDD, Seagate.
- Linux DietPi 4.9.40-v7+ #2 SMP Sat Aug 5 12:35:22 UTC 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi?
Yes.

#### Expected behaviour:
A fresh install should have a version of UNRAR installed so that unpacking of files works within SABnzbd. 

#### Actual behaviour:
Unpacking does not work. 

#### Steps to reproduce:
Install the img from http://dietpi.com/downloads/images/, DietPi_v150_RPi-armv6-(Jessie).img onto an SD-card using Etcher. 

Insert into RPi 2.
Boot and connect through SSH. Allow for updating/upgrading (rebooting twice).
Install the software SABnzbd (only pick this).
Wait for it to be done and reboot. 
Go to 192.168.0.100:8080, finish the configuration wizard.
This will be your view: https://i.imgur.com/DYxM9th.png

#### Extra details:
root@DietPi:~# unrar --version
unrar 0.0.1

Invictaz, 2017-08-09:
Unrar was not provided since it comes from the nonfree repo (if I was informed correctly). DietPi should have an option to install it as per user request.

Fourdee, 2017-08-10:
@k3anton 

Thanks for the report, we'll take a look 👍

Fourdee, 2018-04-05:
As of v6.6, RPi's currently install our non-free binary by default:
http://dietpi.com/downloads/binaries/all/unrar_5.3.2-1+deb9u1_armhf.deb

closed 2018-04-05 by Fourdee