MichaIng/DietPi #1107: Can't start MySQL

pidiet, 2017-08-09:
#### Required Information:
- DietPi Version: **154**
- SBC Device: **Orange Pi PC 2**
- Power supply used: **Xunlong 5v 3A**
- SD card used: **Samsung Evo+ 32GB**
- Distro: **Linux hostname 4.10.0-sun50iw2 #1 SMP Tue May 2 12:07:11 BST 2017 aarch64 GNU/Linux**

#### Additional Information
- Software title: **baikal**
- External Storage: USB3 HD 320GB, SMART: Perfect

#### Expected behaviour:
Following the guide to set up baikal: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1529#p1502
I should restart MySQL to be able to use with baikal with the following command:
`service mysql restart`

#### Actual behaviour:
I get the following Error:
`Job for mysql.service failed. See 'systemctl status mysql.service' and 'journalctl -xn' for details.`

Putting in: systemctl status mysql.service -l

`● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: failed (Result: exit-code) since Wed 2017-08-09 23:10:48 CEST; 2min 8s ago
  Process: 4811 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

Aug 09 23:10:48 hostname mysql[4811]: Starting MySQL database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . failed!
Aug 09 23:10:48 hostname systemd[1]: mysql.service: control process exited, code=exited status=1
Aug 09 23:10:48 hostname systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
Aug 09 23:10:48 hostname systemd[1]: Unit mysql.service entered failed state.`

#### Extra details:
I restarted a couple of time and did some search on the internet. I digged out: /var/log/mysql/error.log 

`170809 22:55:07 [Warning] Using unique option prefix myisam-recover instead of myisam-recover-options is deprecated and will be removed in a future release. Please use the full name instead.
170809 22:55:07 [Note] Plugin 'FEDERATED' is disabled.
/usr/sbin/mysqld: Table 'plugin' is read only
170809 22:55:07 [ERROR] Can't open the mysql.plugin table. Please run mysql_upgrade to create it.
170809 22:55:07 InnoDB: The InnoDB memory heap is disabled
170809 22:55:07 InnoDB: Mutexes and rw_locks use GCC atomic builtins
170809 22:55:07 InnoDB: Compressed tables use zlib 1.2.8
170809 22:55:07 InnoDB: Using Linux native AIO
170809 22:55:07 InnoDB: Initializing buffer pool, size = 128.0M
170809 22:55:07 InnoDB: Completed initialization of buffer pool
170809 22:55:07 InnoDB: highest supported file format is Barracuda.
170809 22:55:07  InnoDB: Waiting for the background threads to start
170809 22:55:08 InnoDB: 5.5.57 started; log sequence number 1595695
170809 22:55:08 [Note] Server hostname (bind-address): '127.0.0.1'; port: 3306
170809 22:55:08 [Note]   - '127.0.0.1' resolves to '127.0.0.1';
170809 22:55:08 [Note] Server socket created on IP: '127.0.0.1'.
170809 22:55:08 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'host' is read only`

The read only could be related to a full disk (not the case: diskuse varies from 3% use on / and 14% on HD) and a wrong appamor settings (https://dba.stackexchange.com/questions/49583/mysql-gives-me-cant-open-and-lock-privilege-tables-table-host-is-read-only)

One more weird thing is that my syncthing was also throwing 'read-only' errors and stopped syncing. A reboot fixed this and syncing continued.

I don't know how to debug any deeper. Any help appreciated!!!

pidiet, 2017-08-09:
I checked the rights of the mysql folder inside of dietpi_userdata (which is on the external hd)

> **drwxrwx**--- 4 avahi utempter 4096 Aug  9 23:10 mysql

Fourdee, 2017-08-10:
Incorrectly set permissions, try:
```
chown -R mysql:mysql /mnt/dietpi_userdata/mysql
dietpi-services restart
```

pidiet, 2017-08-10:
Tried this but get still the same error's with the following folder permissions

> drwxrwx--- 4 mysql mysql 4096 Aug 10 12:59 mysql

> Job for mysql.service failed. See 'systemctl status mysql.service' and 'journalctl -xn' for details.

systemctl status mysql.service

> ● mysql.service - LSB: Start and stop the mysql database server daemon
>    Loaded: loaded (/etc/init.d/mysql)
>    Active: failed (Result: exit-code) since Thu 2017-08-10 12:59:54 CEST; 27s ago
>   Process: 4969 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)
> 
> Aug 10 12:59:54 orangeberry2 mysql[4969]: Starting MySQL database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . failed!
> Aug 10 12:59:54 orangeberry2 systemd[1]: mysql.service: control process exited, code=exited status=1
> Aug 10 12:59:54 orangeberry2 systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
> Aug 10 12:59:54 orangeberry2 systemd[1]: Unit mysql.service entered failed state.

 /var/log/mysql/error.log

> 170810 12:59:23 [Warning] Using unique option prefix myisam-recover instead of myisam-recover-options is deprecated and will be removed in a future release. Please use the full name instead.
> 170810 12:59:23 [Note] Plugin 'FEDERATED' is disabled.
> /usr/sbin/mysqld: Table 'plugin' is read only
> 170810 12:59:23 [ERROR] Can't open the mysql.plugin table. Please run mysql_upgrade to create it.
> 170810 12:59:23 InnoDB: The InnoDB memory heap is disabled
> 170810 12:59:23 InnoDB: Mutexes and rw_locks use GCC atomic builtins
> 170810 12:59:23 InnoDB: Compressed tables use zlib 1.2.8
> 170810 12:59:23 InnoDB: Using Linux native AIO
> 170810 12:59:23 InnoDB: Initializing buffer pool, size = 128.0M
> 170810 12:59:23 InnoDB: Completed initialization of buffer pool
> 170810 12:59:23 InnoDB: highest supported file format is Barracuda.
> 170810 12:59:23  InnoDB: Waiting for the background threads to start
> 170810 12:59:24 InnoDB: 5.5.57 started; log sequence number 1595695
> 170810 12:59:24 [Note] Server hostname (bind-address): '127.0.0.1'; port: 3306
> 170810 12:59:24 [Note]   - '127.0.0.1' resolves to '127.0.0.1';
> 170810 12:59:24 [Note] Server socket created on IP: '127.0.0.1'.
> 170810 12:59:24 [ERROR] Fatal error: Can't open and lock privilege tables: Table 'host' is read only
>

Fourdee, 2017-08-10:
@pidiet 

> One more weird thing is that syncthing was also throwing 'read-only' errors and stopped syncing. A reboot fixed this and syncing continued.

Could be related to unstable system (eg: filesystem corruptions due to PSU/SD).

Lets check dmesg for any errors:
```
dmesg | grep mmc
```

pidiet, 2017-08-10:
`dmesg | grep mmc`

 
> [    2.840740] sunxi-mmc 1c0f000.mmc: Got CD GPIO
> [    2.879985] sunxi-mmc 1c0f000.mmc: base:0xffff000008d0b000 irq:7
> [    2.955601] mmc0: host does not support reading read-only switch, assuming write-enable
> [    2.967337] mmc0: new high speed SDHC card at address 59b4
> [    3.030127] mmcblk0: mmc0:59b4 00000 29.8 GiB 
> [    3.031221]  mmcblk0: p1 p2
> [    5.732920] EXT4-fs (mmcblk0p2): mounted filesystem with writeback data mode. Opts: (null)
> [    8.103574] EXT4-fs (mmcblk0p2): re-mounted. Opts: discard

My dietpi_data is on a new external usb3 hard disk

Fourdee, 2017-08-10:
@pidiet 

> USB3 HD 320GB

How it powered, USB host?

> Can't open the mysql.plugin table. Please run mysql_upgrade to create it.

Try:
```
mysql_upgrade
```

pidiet, 2017-08-10:
Yes. HD is USB powered. The Powersupply is 5V 3A which should be more than enough for the OPiPC2 and the 2.5' HD.

`mysql_upgrade`

> Looking for 'mysql' as: mysql
> Looking for 'mysqlcheck' as: mysqlcheck
> Error: Failed while fetching Server version! Could be due to unauthorized access.
> FATAL ERROR: Upgrade failed

pidiet, 2017-08-11:
hmmm.. couldn't get it running!
Switched over to radicale (http://radicale.org/) -> Doesn't need mysql and feels more lightweight. Only the web interface is very minimal.
I will create a software request for this ;)

pidiet, 2017-08-11:
https://github.com/Fourdee/DietPi/issues/1108

kaffeekanne, 2017-11-05:
Baikal is not developed anymore, so removing it from dietpi might be wise: [http://sabre.io/blog/2017/development-on-hold/](url)

Fourdee, 2018-02-22:
Closing due to:
- Outdated < v6.0
- MariaDB is now default
- OPi images/support have been dropped.

closed 2018-02-22 by Fourdee