MichaIng/DietPi #1108: Software Request:

pidiet, 2017-08-11:
# Radicale

**A Free and Open-Source CalDAV and CardDAV Server**

## About Radicale

Radicale is a small but powerful CalDAV (calendars, todo-lists) and CardDAV (contacts) server, that:

>     Shares calendars through CalDAV, WebDAV and HTTP.
>     Shares contacts through CardDAV, WebDAV and HTTP.
>     Supports events, todos, journal entries and business cards.
>     Works out-of-the-box, no installation nor configuration required.
>     Can warn users on concurrent editing.
>     Can limit access by authentication.
>     Can secure connections.
>     Works with many CalDAV and CardDAV clients.
>     Is GPLv3-licensed free software. 

http://radicale.org/

kaffeekanne, 2017-11-05:
Is radicale still considered? Development of Baikal is halted, and radicale seems to be the alternative with low resources needed. #1107

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

d4f2, 2018-05-03:
please add Radicale to dietpi

pedrom34, 2018-05-04:
Well, I looked around a bit, and there's a package in debian. So with 
``shell
apt-get install radicale``
I made it work.

baldpitter, 2018-05-04:
Sure, installing the debian package is an option. However, I too think that dietpi should support radicale natively. That would be a great addition!

STPKITT, 2019-07-11:
I can only agree. While Baikal is back into development Radicale should be offered as a more slim & robust alternative since it works without needing a database.

iraklis10, 2024-08-31:
Another vote for radicale