MichaIng/DietPi #1109: Ubuntu Mate does not auto start

fredjedimaster, 2017-08-14:
Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device: Orange PI Zero 256mb
- Power supply used: 5v 2A Samsung Power Supply
- SD card used: Sandisk ultra 16gb Class 10
- Distro (EG: Jessie) | ```uname -a```

#### Expected behaviour:
Ubuntu Mate should be starting automatically after configuring the optionon dietpi-autostart. Its already configured with the option: 2 (desktop) and starts normally when type startx on command line, bit i dont want to use the keyboard (webcam server for security home).

The project is:

Power on opizero
Start dietpi
Start Ubuntu Mate
Start motion service

Acess the webcam imagem on mobile phone...PC...etc...

#### Actual behaviour:

The system does not start automatically, only when i type startx command.

#### Steps to reproduce:
Flash dietpi opizero imagem
Install ubuntu mate desktop software

Try configure dietpi-autostart

#### Did you submit a ```dietpi-bugreport```?
No. How to?

#### Extra details:
There's two more users in the dietpi forum with the same problem on orange PI boards.

Fourdee, 2017-08-14:
@fredjedimaster 

Thanks for the report 👍 

We'll investigate.

Fourdee, 2017-08-17:
XFCE report with OPi Plus: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2138&p=8271#p8265

fredjedimaster, 2017-08-17:
Thanks for the quick answer...

Fourdee, 2017-08-18:
Notes:
- tty check is no longer valid ```if [ "$(tty)" = "/dev/pts/0" ]; then```
- most likely kernel/armbian update, tty is now ```/dev/tty1``` as per all other boards.

Fourdee, 2017-08-18:
Fixed for v155. Can be patched now with:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/98d348527baa0b7c92557254898c40845b22328e/dietpi/login -O /DietPi/dietpi/login
reboot
```

closed 2017-08-18 by Fourdee

fredjedimaster, 2017-08-18:
> Enviada em: sexta-feira, 18 de agosto de 2017 07:42
>
>
>
> Closed #1109 <https://github.com/Fourdee/DietPi/issues/1109> .
>
>
> ---
> Este email foi escaneado pelo Avast antivírus.
> https://www.avast.com/antivirus

Thanks. It worked. Started on mate automatically.