MichaIng/DietPi #1110: OrangePi Zero 2Plus H5 boot hang

jiangyanfeng, 2017-08-18:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
boot successfuly

#### Actual behaviour:
boot hang at "fuse init (API version 7.26)" and a few minutes later, "armbian GNU/Linux 8 DietPi tty1", and still cannot login 

#### Steps to reproduce:
1、download and uncompress DietPi_v150_OrangePiZero2PlusH5-arm64-(Jessie).img from dietpi.com
2、format the tf card (the img file has been check the hash value)
3、write DietPi_v150_OrangePiZero2PlusH5-arm64-(Jessie).img to the tf card
4、boot the OrangePi Zero 2Plus H5 from tf

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-08-18:
@jiangyanfeng 

Thanks for the report 👍 

We'll check.

jiangyanfeng, 2017-08-21:
I found a error message in the boot log: 

systemd[1]: Cannot add dependency job for unit display-manager.service, ignoring: Unit display-manager.server failed to load: No such file or directory.

jiangyanfeng, 2017-08-21:
I take a photo for the boot screen:
http://www.donghootech.com/ueditor/php/upload/image/20170820/1503284747878241.jpg

Fourdee, 2017-08-22:
@jiangyanfeng 
> display-manager.service

Thats not one of our services, can you check status?

```
systemctl status display-manager.service -l
```

jiangyanfeng, 2017-08-23:
@Fourdee As above said, I cannot boot into console cmd line, so I cannot run the cmd to check the service status. But I think the display-manager.service is not the reason cause boot hang.

Fourdee, 2017-08-26:
🈴 Confirmed. Occurs after apt-get upgrade (kernel upgrade)

Fourdee, 2017-08-26:
Moved out of milestone, this is a kernel related issue (we use ARMbian) outside of our control.

Fourdee, 2018-02-22:
Closing as OPi support has been dropped.

closed 2018-02-22 by Fourdee