MichaIng/DietPi #1112: Octoprint auto update

qpens, 2017-08-20:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version v154
- SBC Device Odroid C2
- Power supply used 2.5A
- SD card used Lexor 633x
- Distro Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
Maintenance RCs Upgrade to 1.3.5 RC2

#### Actual behaviour:
Updating, please wait.++++++++++++++++++++++++++++++++++Now updating OctoPrint to 1.3.5rc2++++++++++++++++++++++++++++++++++/usr/bin/python "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch=rc/maintenance --force=true "/home/dietpi/" 1.3.5rc2Python executable: '/usr/bin/python'>>> Running: git diff --shortstat> git diff --shortstatNot a git repositoryTo compare two paths outside a working tree:usage: git diff [--no-index] <path> <path>Traceback (most recent call last):File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>main()File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 309, in mainupdate_source(git_executable, folder, args.target, force=args.force, branch=args.branch)File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 204, in update_sourceif _rescue_changes(git_executable, folder):File "/usr/local/lib/python2.7/dist-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 180, in _rescue_changesraise RuntimeError("Could not update, \"git diff\" failed with returncode {}".format(returncode))RuntimeError: Could not update, "git diff" failed with returncode 129The update did not finish successfully. Please consult the log for details

Fourdee, 2017-08-22:
@qpens 

Thanks for the report 👍 

Looks like a possible bug with their update script:
```
Python executable: '/usr/bin/python'>>> Running: git diff --shortstat> git diff --shortstatNot a git repositoryTo compare two paths outside a working tree:usage: git diff [--no-index] Traceback (most recent call last):
```

We'll run some tests and try to replicate.

Fourdee, 2017-10-14:
Confirmed, when using another update branch, other than default.

Due to DietPi only cloning with ```depth=1``` (attempt to keep install lightweight):
```
git clone "$INSTALL_URL_ADDRESS" --depth=1
```

I'll remove the ```depth=1``` for v157. Fresh installs can then switch to any update branch:
[image]

Fourdee, 2017-10-14:
Resolved for v157.

closed 2017-10-14 by Fourdee