MichaIng/DietPi #1113: Odroid XU4 Install issue

gregb49, 2017-08-20:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device Odroid XU4
- Power supply used 5v 4.0A CWT AC Adaptor?
- SD card used Sandisk ultra 32GB - new
- Distro cat Linux DietPi 3.10.105+ #1 SMP PREEMPT Mon Aug 7 10:09:24 CEST 2017 armv71 GNU/Linux
#### Additional Information (if applicable):
- Software title? NK
- Can this issue be replicated on a fresh installation of DietPi? This is a fresh install

#### Expected behaviour:
 Software should be installing using initialisation sequence.

#### Actual behaviour:
 Error Code 100 Apt-get failed

#### Steps to reproduce:
 NK

#### Did you submit a ```dietpi-bugreport```?
 This is it

#### Extra details:

contents of ?var/log/dietpisoftware_apt.log
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

Fourdee, 2017-08-25:
@gregb49 
Thanks for the report 👍 

Please run the following, check for any errors on screen.
```
apt-get -f install
```

If no errors, please try installing the software title again.

Fourdee, 2018-02-22:
Closing due to no response/outdated.

closed 2018-02-22 by Fourdee