MichaIng/DietPi #1114: DietPi-Services | Feature Req | Make DietPi-Services Stop accept service name arguments

ReanimationXP, 2017-08-21:
Self-explanatory. Would like to be able to use this akin to net start / net stop, and use the commands on individual services. 

`dietpi-services stop apache`
`dietpi-services restart apache`
`dietpi-services disable apache` (would prevent autostart on boot)

Fourdee, 2017-08-21:
@ReanimationXP 
```
dietpi-services stop apache
```

Great idea 👍 

> ```dietpi-services disable apache```

Yep, we can do that. Although we'd need to mask the service as all services are disabled by default, to allow ```dietpi-services``` to take control.

Invictaz, 2017-08-21:
Interesting idea

ReanimationXP, 2017-08-21:
> we'd need to mask the service as all services are disabled by default, to allow dietpi-services to take control.
Mask which service? I don't follow.

Fourdee, 2017-12-10:
I added the ability to ```mask/unmask``` services using:
```dietpi-services mask servicename```

https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-services#L16

This will prevent it from running. We cant use ```disable``` as DietPi relies on this to take over control from SystemD.

Completed.

Fourdee, 2017-12-10:
Reopened to consider:
> dietpi-services stop apache
dietpi-services restart apache
dietpi-services disable apache

Fourdee, 2017-12-10:
[image]

ReanimationXP, 2017-12-10:
Thanks for implementing this. I'd like to suggest the term `mask` be changed to something else more akin to `disable`.  Without reading this thread, even with the help above, I'm completely unclear on what that functionality would do.

Fourdee, 2017-12-10:
@ReanimationXP 
> I'd like to suggest the term mask be changed to something else more akin to disable

Agree 👍

ReanimationXP, 2017-12-10:
I assume what is happening is all dietpi-services are separate from the list of systemd services, and thus you are enabling/disabling within the context of dietpi, correct? If so, I think disable still applies, just perhaps with an asterisk that `enable`, `disable`, and any other related commands cannot be used on systemd services without importing/converting them into the dietpi context.

Fourdee, 2017-12-10:
@ReanimationXP 

- ```enable/disable/stop/start``` will work for any SystemD service.
- Items marked (ALL), will only effect services known to DietPi: https://github.com/Fourdee/DietPi/issues/1114#issuecomment-350516067

ReanimationXP, 2017-12-10:
I'm still confused then. I was drawing my conclusions from this comment:

> This will prevent it from running. We cant use disable as DietPi relies on this to take over control from SystemD.

Back to wording, `activate`/`deactivate` or `mark`/`unmark` might be a good alternatives to mask/unmask.  Only problem there is `active` might infer running state.

ReanimationXP, 2017-12-10:
I see what you're saying now.. that disable when applied to a systemd service would take a different action.  Applied to a dietpi-service, it would be functionally the same as `mask`.

Fourdee, 2017-12-10:
[image]

ReanimationXP, 2017-12-10:
erhm, k now confused again. lol.  Assuming there was already a cron service in systemd, what happened to it when you symlink it to dev/null?

Fourdee, 2017-12-10:
@ReanimationXP 

Disables it, prevents it from running.

ReanimationXP, 2017-12-10:
Well, right, but was it not overwritten?  Where did it go should they want to add it back manually?

Fourdee, 2017-12-10:
@ReanimationXP 

> https://askubuntu.com/questions/816285/what-is-the-difference-between-systemctl-mask-and-systemctl-disable
When you mask a service, a symlink is created from /etc/systemd/system to /dev/null, leaving the original unit file elsewhere untouched. When you unmask a service the symlink is deleted.

Fourdee, 2017-12-10:
Marking as completed.

closed 2017-12-10 by Fourdee

jkossis, 2018-08-06:
@Fourdee 

For clarity, if I wanted to disable a diet-pi service from starting at boot (such as transmission-daemon),
do I do it through systemctl, or through dietpi-services? If the later, do I need to run some combination of mask/unmask as well as disable? Just a little confused. Could be my lack of understanding here.

I appreciate the help.

Fourdee, 2018-08-06:
@jkossis 

> For clarity, if I wanted to disable a diet-pi service from starting at boot (such as transmission-daemon),
do I do it through systemctl, or through dietpi-services? If the later, do I need to run some combination of mask/unmask as well as disable? Just a little confused. Could be my lack of understanding here.

By default DietPi-Services disables services which allows it to take manual start/stop control of.

To disable DietPi-Services from controlling the service, you need to either:
- Mask service ```systemctl mask servicename```
- ```dietpi-services disable servicename```. This is the same as running the command above.

MichaIng, 2018-08-06:
@jkossis 
And as masking does not work, if the original .service file is placed within /etc/systemd/system/, to completely disable those services, you need to remove the related files or better move them to a backup location.

This is the opened issue: https://github.com/Fourdee/DietPi/issues/1809

jkossis, 2018-08-06:
@MichaIng 

It seems that the issue linked is for manually created services.

@Fourdee

I don't want to disable dietpi-services from controlling transmission-daemon. I am looking for a way in which I can tell dietpi-services to not auto start transmission-daemon, but rather, wait for me to manually start it.

To achieve this, do I need to relinquish the control from dietpi-services to systemctl?

MichaIng, 2018-08-06:
@jkossis 
> It seems that the issue linked is for manually created services.

Jep mostly, but indeed also some software installs place their service units into that folder.

If I see it right, then currently dietpi-services handles services in every case, if they are fount either within /lib/systemd/system of /etc/systemd/system.
Thus you need to remove the service name from the script for this currently:
`sed -i '/transmission-daemon/d' /DietPi/dietpi/dietpi-services`

After every dietpi-update this needs to be redone.

Makes sense we think about a user friendly solution to exclude scripts from dietpi-services (and systemd), if desired.

Fourdee, 2018-08-08:
@MichaIng 

> Makes sense we think about a user friendly solution to exclude services from dietpi-services (and systemd), if desired.

Yep, we have an include, not exclude. I'll see about adding that in.

Fourdee, 2018-08-08:
@jkossis 

Please update the script:
```
wget https://github.com/Fourdee/DietPi/raw/6444d764b4d0a9bce099fecf9452370b32082ed4/dietpi/dietpi-services -O /DietPi/dietpi/dietpi-services
```

Run DietPi services at least once to generate the new include/exlude file:
```
dietpi-services
```

Edit the file ```/DietPi/dietpi/.dietpi-services_include_exclude``` to add include/excludes. All info is in there.