MichaIng/DietPi #1115: Netdata 1.7

Kapot, 2017-08-22:
Seems NetData is running on v1.7.0 (https://github.com/firehol/netdata/releases) but I cant seems to get it updated using dietpi-update. Running v154.

R1CH-T, 2017-08-25:
Hi, @Kapot,

I have just looked into this and it seems that DietPi is currently hosting it's own repository for a number of NetData `.deb` installers, for this application.

DietPi has evolved to work seemlessly and in the same way for a variety of Single Board Computers and aims to keep the experience the same for all, hence why it hosts these older files.

This obviously needs to be looked at, tested and the installers updated to the latest versions, [but other things are being worked on at the moment](https://github.com/Fourdee/DietPi/milestone/58?closed=1) and it currently 'just works'.

Please let people know about any new functionality, or whatever, that you are excited about in newer versions of NetData, on the [DietPi Forums](http://dietpi.com/phpbb/viewforum.php?f=5) and you may gain traction for your request.

https://github.com/firehol/netdata/releases

http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software (Search for "netdata" in the file in order to see how NetData is installed in DietPi: It _should_ be an easy matter to replace the packages.)

Please look into [Package Management](https://en.wikipedia.org/wiki/List_of_software_package_management_systems) to see why this sort of stuff is a problem and why DietPi might be lagging behind in some areas and ahead in others.

Thanks,

Rich.

MichaIng, 2018-01-31:
Self reminder: Check installation of current Debian package from pool: http://ftp.de.debian.org/debian/pool/main/n/netdata/

Fourdee, 2018-02-22:
Netdata v1.9 as of DietPi v6.2.

closed 2018-02-22 by Fourdee