MichaIng/DietPi #1118: Question: Automated installation

douglasmiranda, 2017-08-25:
Is there a way to automatize the installation? 

If I want to build a cluster, and DietPi needs to be installed in several boards, how should I proceed?

R1CH-T, 2017-08-25:
Hi,

Take a look at https://github.com/Fourdee/DietPi/blob/master/dietpi.txt.

This file can be modified to automate various installation options, software selections etc.

It should be modified after you write DietPi to the SD card(s) and before you insert it into your board(s) and boot up.

Hope this helps.

douglasmiranda, 2017-08-25:
Awesome!!

(But shame on me, how could I not see that file..)

R1CH-T, 2017-08-25:
Cool!

Post your links and experiences to the [DietPi Forums](http://dietpi.com/phpbb/viewforum.php?f=5).

> (But shame on me, how could I not see that file..)

It happens to all of us. I expect that you read the relevant instructions early on and forgot the bits you needed to know, several hours later!

douglasmiranda, 2017-08-25:
I think my username wasn't approved on the DietPi Forum yet =/ douglasmirandasilva@gmail.com

Fourdee, 2017-08-26:
@douglasmiranda 
Unable to find user with your email address. Most likely triggered as spam. 

Please can you recreate the account and provide username. I'll active.

MichaIng, 2018-01-31:
Close, as question was answered.

closed 2018-01-31 by MichaIng