MichaIng/DietPi #1119: Error | syslog/ipv6 | Couldn't write '1' to 'net/ipv6/conf/all/disable_ipv6' ...

MichaIng, 2017-08-25:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```154```
- SBC Device (EG: RPi 3)? RPi 2
- Power supply used (EG: 5v 1A RAVpower)? 5V 2A
- SD card used (EG: Sandisk ultra)? Kingston SDCA3/64GB
- Distro (EG: Jessie) | ```Stretch```

#### Additional Information (if applicable):
- Software title? part of dietpi-config
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:

No error

#### Actual behaviour:

```
Aug 24 18:00:12 micha systemd-sysctl[136]: Couldn't write '1' to 'net/ipv6/conf/all/disable_ipv6', ignoring: No such file or directory
Aug 24 18:00:12 micha systemd-sysctl[136]: Couldn't write '1' to 'net/ipv6/conf/default/disable_ipv6', ignoring: No such file or directory
Aug 24 18:00:12 micha systemd-sysctl[136]: Couldn't write '1' to 'net/ipv6/conf/lo/disable_ipv6', ignoring: No such file or directory
```

#### Steps to reproduce:

1. disable IPv6 via dietpi-config
2. reboot
3. Have a look into `/var/log/syslog`

#### Did you submit a ```dietpi-bugreport```?

I left it out this time ;).

#### Extra details:

- By disabling IPv6 via dietpi-config, the ipv6 kernel module will be blacklisted: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L3096-L3128
- Therefore the module will be not loaded anymore and the path `/proc/sys/net/ipv6/...` does not exist anymore after reboot.
- By `net.ipv6.conf.XXX.disable_ipv6` inside `/etc/sysctl.conf` sysctl is now looking and missing exactly this paths 😄 .

#### How to fix:
- Remove all sysctl.conf manipulations from https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L3096-L3128 as creating/removing the blacklisting of the module does everything necessary.
- Also remove the entries by [patch_file](https://github.com/Fourdee/DietPi/blob/testing/dietpi/patch_file#L2678).
- Also remove the entries from [PREP_SYSTEM_FOR_DIETPI.sh](https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L348-L353) as they also default to **0** (enable ipv6) in case the module is active.

#### Additional information
- https://askubuntu.com/questions/748636/issues-with-disabling-ipv6-on-ubuntu-14-0-4
- https://en.wikipedia.org/wiki/Sysctl

I can create the PR, just want to be sure that I didn't miss something.

Fourdee, 2017-08-25:
@MichaIng 

> I can create the PR, just want to be sure that I didn't miss something.

Looks good, please send PR if possible 👍

MichaIng, 2017-10-29:
@Fourdee This can be closed by the way 😉.

closed 2017-10-30 by Fourdee