MichaIng/DietPi #1121: Kernel panic on install after updates

Elpadreloco, 2017-08-26:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | v150
- SBC Device RPi3
- Power supply used 5V 3A Akru
- SD card used Kingston SDHC 16Gb
- Distro EG: Jessie

#### Expected behaviour:
Dietpi starting for the first time

#### Actual behaviour:
Error code : end kernel panic - not syncing - Attempted to kill init ! exit code 0x00007f00

#### Steps to reproduce:
Flash the fresh downloaded Dietpi image on SD card, boot. After fist steps, Dietpi lanch automatic update. From v150 to 153, nothing wrong but when it update to 154, it reboot and display the error message above.

#### Did you submit a ```dietpi-bugreport```?
No, can't

#### Extra details:

Fourdee, 2017-08-26:
@Elpadreloco 
Thanks for the report 👍 

We were unable to replicate this in our tests. Please verify SD card and PSU are stable.

Elpadreloco, 2017-08-26:
Thank you for the response. SD card works perfectly with Raspbian, OpenSuse or Recal. Same thing with PSU.

Fourdee, 2017-08-26:
@Elpadreloco 

> works perfectly with Raspbian, OpenSuse or Recal

These are preinstalled images that do very little during 1st run. DietPi is different, it does alot of filesystem IO during 1st run to update and patch system. 
If the device, SD card and/or PSU are unstable, it will fail during this early stage. Please check your hardware.

D3add3d, 2017-09-10:
I also have this issue, SD card is tested and formatted Class 10, 16GB Kingston, RPi 1 B, power supplied from powered D-Link USB hub that can supply 2.1A@5V to 2 powered ports, from experience I can tell this setup is very stable...
I've got the same exact code, watching the install I spotted an error message saying that the compressed file of the kernel violated the format

D3add3d, 2017-09-10:
Ok... that was some other error, here is a video showing the failed extraction of kernel (this is my second try - reformated, reflashed): https://youtu.be/HLAsf8SIGdk //sorry about the music, YouTube does not allow the sound to be simply disabled and there was quite a bit of background noise

Fourdee, 2017-09-10:
@D3add3d 
[image]
?

D3add3d, 2017-09-10:
Sorry, my bad, forgot to switch it from private to unlisted, should be fixed.now

Fourdee, 2017-09-10:
@Elpadreloco 

Thanks, looks like the unzip is failing. 

Lets try and run it manually:
```
wget http://dietpi.com/downloads/binaries/rpi/dietpi_rpi_kernel_4.9.zip -O package.zip
unzip -o package.zip -d /
```

Please check for any errors.

D3add3d, 2017-09-10:
I... can't? There is no way, the autoupdate runs automatically and when it reboots the kernel panics

Fourdee, 2017-09-10:
@D3add3d 

You can press CTRL+C to terminate the current program.

Do this during kernel extraction (as per the video).

D3add3d, 2017-09-10:
ok, I will try it later

D3add3d, 2017-09-11:
@Fourdee ok,  I ran it manually, I got the same output, many errors at the beginning and then inflating and create

closed 2018-02-22 by Fourdee