MichaIng/DietPi #1122: Openssh doesn't autoinstall when set default at "dietpi.txt" before first run

joaofl, 2017-08-28:
When Openssh is set to install by default (to enable sftp) at "dietpi.txt" before first run, it does not get installed.

After I edit the dietpi.txt correspondingly and  after first run setup, dropbear is still the one up running. 
However, I go to dietpi-software menu, and openssh is there selected as preferred. So I have to select dropbear again, let it reisntall it, then go back there and switch again to openssh, so only then It removes dropbear and install openssh.

`#DietPi-Software Choice System
#SSH Server Selection:
#	0=none
#	-1=dropbear
#	-2=opensshserver
AUTO_DietpiSoftware_SSHServerIndex=-2`

Fourdee, 2017-08-28:
@joaofl 

Thanks for the report 👍 

> I go to dietpi-software menu, and openssh is there selected as preferred. 

I've had this a few times, however, seems intermittent and unable to find the cause so far.

Fourdee, 2017-11-10:
Notes:
- Seems to occur 100% of time with RPi + Stretch image.

Fourdee, 2017-12-05:
Notes v2:
```
root@DietPi:~# cat /etc/dietpi/logs/dietpi-firstrun-setup.log
 [Info] Checking RootFS R/W access. Please wait...
 [Ok] RootFS R/W access.
 [Info] Initializing database, please wait...
DEBUG: AUTOINSTALL_SSHINDEX 0
DEBUG: INDEX_SSHSERVER_CURRENT -1
DEBUG: INDEX_SSHSERVER_TARGET -1
 [Info] Reading database, please wait...
DEBUG: AUTOINSTALL_SSHINDEX 0
DEBUG: INDEX_SSHSERVER_CURRENT -2
DEBUG: INDEX_SSHSERVER_TARGET -2
```

Fourdee, 2017-12-07:
Due to outdated code, when ```dietpi-uninstall``` existed as a separate script to ```dietpi-software```

Fourdee, 2017-12-07:
Test:
- 🈯️ Odroid C2 Stretch

Fourdee, 2017-12-07:
Failed on:
- 🈴 RPi 3

Fourdee, 2017-12-08:
@k-plan confirms issue occurs with use of dietpi-automation:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=2542&p=9846#p9844

k-plan, 2017-12-08:
@Fourdee 

>with use of dietpi-automation

only for the sake of completeness: 
- I have done this by DietPi inital setup assistent and choose OpenSSH by hand, because I want to see what's happend. But I can't see any messages or get logs. And it was on Jessie.

k-plan, 2017-12-08:
Testing:

- fresh image auto-updated => v158

- first change to 'Full:  Logrotate and Rsyslog" => installed => auto-reboot

- switching now from `Dropbear` => `OpenSSH` => auto installed / uninstall => auto-reboot
works and got some logs:
```
root@RPi-image:~# cat /var/log/dietpi-software_apt.log
Preconfiguring packages ...
Selecting previously unselected package libwrap0:armhf.
(Reading database ... 19270 files and directories currently installed.)
Preparing to unpack .../libwrap0_7.6.q-25_armhf.deb ...
Unpacking libwrap0:armhf (7.6.q-25) ...
Selecting previously unselected package openssh-client.
Preparing to unpack .../openssh-client_1%3a6.7p1-5+deb8u3_armhf.deb ...
Unpacking openssh-client (1:6.7p1-5+deb8u3) ...
Selecting previously unselected package openssh-sftp-server.
Preparing to unpack .../openssh-sftp-server_1%3a6.7p1-5+deb8u3_armhf.deb ...
Unpacking openssh-sftp-server (1:6.7p1-5+deb8u3) ...
Selecting previously unselected package openssh-server.
Preparing to unpack .../openssh-server_1%3a6.7p1-5+deb8u3_armhf.deb ...
Unpacking openssh-server (1:6.7p1-5+deb8u3) ...
Processing triggers for systemd (215-17+deb8u7) ...
Setting up libwrap0:armhf (7.6.q-25) ...
Setting up openssh-client (1:6.7p1-5+deb8u3) ...
Setting up openssh-sftp-server (1:6.7p1-5+deb8u3) ...
Setting up openssh-server (1:6.7p1-5+deb8u3) ...
Creating SSH2 RSA key; this may take some time ...
2048 62:c7:9f:98:cb:ea:7b:77:69:16:ad:e4:44:36:ed:75 /etc/ssh/ssh_host_rsa_key.pub (RSA)
Creating SSH2 DSA key; this may take some time ...
1024 5e:15:76:f9:59:7f:73:82:bc:85:cc:e8:82:62:47:49 /etc/ssh/ssh_host_dsa_key.pub (DSA)
Creating SSH2 ECDSA key; this may take some time ...
256 71:86:27:db:c6:1b:c7:d8:f8:74:3c:88:7c:cd:46:3a /etc/ssh/ssh_host_ecdsa_key.pub (ECDSA)
Creating SSH2 ED25519 key; this may take some time ...
256 ee:b1:94:4e:f5:a2:a1:0d:57:33:7d:dd:50:eb:03:ba /etc/ssh/ssh_host_ed25519_key.pub (ED25519)
Processing triggers for libc-bin (2.19-18+deb8u10) ...
Processing triggers for systemd (215-17+deb8u7) ...

```

```
root@RPi-image:~# cat /var/log/apt/history.log

Start-Date: 2017-12-08  21:39:31
Commandline: apt-get install -y -qq --force-yes openssh-server --no-install-recommends
Install: openssh-sftp-server:armhf (6.7p1-5+deb8u3, automatic), openssh-server:armhf (6.7p1-5+deb8u3), libwrap0:armhf (7.6.q-25, automatic), openssh-client:armhf (6.7p1-5+deb8u3, automatic)
End-Date: 2017-12-08  21:40:25

Start-Date: 2017-12-08  21:40:59
Commandline: apt-get purge -y dropbear*
Purge: dropbear:armhf (2014.65-1+deb8u2)
End-Date: 2017-12-08  21:41:07

```

- switching back to `Dropbear` and  `DietPi-Ramlog #2 ` => auto installed / uninstall => auto-reboot

```
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libwrap0 openssh-sftp-server
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  openssh-server*
0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
After this operation, 820 kB disk space will be freed.
(Reading database ... 19272 files and directories currently installed.)
Removing openssh-server (1:6.7p1-5+deb8u3) ...
Purging configuration files for openssh-server (1:6.7p1-5+deb8u3) ...

```
works but now where are some issue:

```
root@RPi-image:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... The following packages were automatically installed and are no longer required:
  libwrap0 openssh-sftp-server
Use 'apt-get autoremove' to remove them.
```

Fourdee, 2017-12-08:
Think i've found the issue. Appears ```.installed``` and a few other files are not being cleared during ```finalize``` in our images:
[image]

k-plan, 2017-12-09:
Is this cause the problem?

`aSOFTWARE_INSTALL_STATE[0]=2`

What can we do? All my images are ready and uploaded. Can you clear the file or should i do?

We only need this:

[image]
Correct?

Fourdee, 2017-12-09:
@k-plan 

> aSOFTWARE_INSTALL_STATE[0]=2

This is the OpenSSH client, which is also installed with OpenSSH server. So we mark is as installed aswell.

> We only need this:

Yep, thats how it should look 👍

Fourdee, 2017-12-09:
#### dietpi.txt automation tests for SSH server choice OpenSSH:
- 🈯️ RPi 3 Stretch

Fourdee, 2017-12-09:
Appears the left over ```.installed``` file on some images may of been the cause. We now reset choice system installed states to defaults during 1st run to overcome this.

Completed.

closed 2017-12-09 by Fourdee

k-plan, 2017-12-09:
@Fourdee 

- cleared all bother files in my build images for now by hand.
But hashes don't fit any more. 😢 
So will repack and upload this tomorrow. Must get sleep now.

Fourdee, 2017-12-09:
@k-plan 

The following image is correct: https://github.com/Fourdee/DietPi/issues/1122#issuecomment-350405872

Which images were not the case?