MichaIng/DietPi #1123: Uboot in meveric repo breaks system

studioai, 2017-08-28:
#### Required Information:
- DietPi Version | 154
- SBC Device (EG: RPi 3)? XU4 
- Power supply used (EG: 5v 1A RAVpower)? 5v6a
- SD card used (EG: Sandisk ultra)? Transcend
- Distro (EG: Jessie) | ```uname -a```? Debian

After Updating uboot package in meveric repo, XU4 doesn't boot.

I tried same thing on the fresh install of dietpi 154 image and result is same.

Unfortunately I overwrote my backup after uboot update.
How can I recover my xu4?

any information will be appreciated!

```
root@DietPi:~# apt-get update && apt-get upgrade
Hit http://security.debian.org jessie/updates InRelease
Hit http://security.debian.org jessie/updates/main armhf Packages                      
Hit http://fuzon.co.uk all InRelease                                         
Hit http://security.debian.org jessie/updates/contrib armhf Packages                   
Hit http://fuzon.co.uk jessie InRelease                                                
Hit http://security.debian.org jessie/updates/non-free armhf Packages             
Hit http://security.debian.org jessie/updates/contrib Translation-en                   
Hit http://fuzon.co.uk all/xu3 armhf Packages                                          
Hit http://security.debian.org jessie/updates/main Translation-en                      
Hit http://fuzon.co.uk all/main armhf Packages                                         
Hit http://security.debian.org jessie/updates/non-free Translation-en                  
Hit http://fuzon.co.uk all/testing armhf Packages                                      
Ign http://ftp.debian.org jessie InRelease                              
Hit http://fuzon.co.uk jessie/main armhf Packages
Hit http://ftp.debian.org jessie-updates InRelease             
Hit http://fuzon.co.uk jessie/backports armhf Packages                  
Hit http://ftp.debian.org jessie-backports InRelease           
Hit http://ftp.debian.org jessie Release.gpg                            
Get:1 http://ftp.debian.org jessie-updates/main armhf Packages/DiffIndex [8,392 B]
Hit http://ftp.debian.org jessie-updates/contrib armhf Packages
Get:2 http://ftp.debian.org jessie-updates/non-free armhf Packages/DiffIndex [736 B]
Hit http://ftp.debian.org jessie-updates/contrib Translation-en
Get:3 http://ftp.debian.org jessie-updates/main Translation-en/DiffIndex [3,196 B]
Get:4 http://ftp.debian.org jessie-updates/non-free Translation-en/DiffIndex [736 B]
Get:5 http://ftp.debian.org jessie-backports/main armhf Packages/DiffIndex [27.8 kB]
Get:6 http://ftp.debian.org jessie-backports/contrib armhf Packages/DiffIndex [25.3 kB]
Get:7 http://ftp.debian.org jessie-backports/non-free armhf Packages/DiffIndex [8,530 B]
Get:8 http://ftp.debian.org jessie-backports/contrib Translation-en/DiffIndex [7,468 B]
Get:9 http://ftp.debian.org jessie-backports/main Translation-en/DiffIndex [27.8 kB]   
Get:10 http://ftp.debian.org jessie-backports/non-free Translation-en/DiffIndex [17.2 kB]
Hit http://ftp.debian.org jessie Release                                               
Hit http://ftp.debian.org jessie/main armhf Packages                                   
Hit http://ftp.debian.org jessie/contrib armhf Packages                                
Hit http://ftp.debian.org jessie/non-free armhf Packages                               
Hit http://ftp.debian.org jessie/contrib Translation-en                                
Hit http://ftp.debian.org jessie/main Translation-en                                   
Hit http://ftp.debian.org jessie/non-free Translation-en                               
Ign http://fuzon.co.uk all/main Translation-en_GB                                      
Ign http://fuzon.co.uk all/main Translation-en                                         
Ign http://fuzon.co.uk all/testing Translation-en_GB                                   
Ign http://fuzon.co.uk all/testing Translation-en                                      
Ign http://fuzon.co.uk all/xu3 Translation-en_GB                                       
Ign http://fuzon.co.uk all/xu3 Translation-en                                          
Ign http://fuzon.co.uk jessie/backports Translation-en_GB                              
Ign http://fuzon.co.uk jessie/backports Translation-en                                 
Ign http://fuzon.co.uk jessie/main Translation-en_GB                                   
Ign http://fuzon.co.uk jessie/main Translation-en                                      
Fetched 127 kB in 16s (7,941 B/s)                                                      
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  uboot
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 261 kB of archives.
After this operation, 344 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://fuzon.co.uk/meveric/ all/xu3 uboot armhf 20170824-9+deb8 [261 kB]
Fetched 261 kB in 2s (91.4 kB/s)              
(Reading database ... 14623 files and directories currently installed.)
Preparing to unpack .../uboot_20170824-9+deb8_armhf.deb ...
Unpacking uboot (20170824-9+deb8) over (20170522-5+deb8u2) ...
Setting up uboot (20170824-9+deb8) ...
30+1 records in
30+1 records out
15616 bytes (16 kB) copied, 0.0783156 s, 199 kB/s
28+1 records in
28+1 records out
14592 bytes (15 kB) copied, 0.0673221 s, 217 kB/s
1209+1 records in
1209+1 records out
619462 bytes (619 kB) copied, 3.90051 s, 159 kB/s
512+0 records in
512+0 records out
262144 bytes (262 kB) copied, 1.91844 s, 137 kB/s
32+0 records in
32+0 records out
16384 bytes (16 kB) copied, 0.185056 s, 88.5 kB/s
root@DietPi:~# reboot
```

Fourdee, 2017-08-28:
Hi,

Thanks for the report 👍 

I'll discuss with Meveric to see if its a local issue. In the mean time, the only solution would be to hold the ```uboot``` package, prior to ```apt-get upgrade```:
```
apt-mark hold uboot
```

studioai, 2017-08-28:
Is it known issue? If it is, Where can I find the information about it? (like forum thread)

nphil, 2017-08-29:
Had the same thing happen to me, lost my set up. Is there any way you could have DietPi not do a uboot or firmware upgrade automatically when doing apt-get upgrade? I sometimes just want to update my software packages which are safe and won't bork up the entire system.

studioai, 2017-08-29:
@nphil

As Fourdee said

'apt-mark hold uboot' will prevent uboot upstaged via apt-get.

studioai, 2017-08-30:
Recovering was so easy If you have backup!

uboot installation doesn't touch boot or root partition. It only modifies boot records on the disk.

So, Format your SD card with fresh image of dietpi and restore the backup.

After that it boots up normally.

rncwnd79, 2017-08-30:
I got another uboot update yesterday. Does this resolve the issue? (Didn't reboot after the last 2 uboot updates).
Or do I have to downgrade? If so, to which version?

Fourdee, 2017-08-30:
@rncwnd79 

> I got another uboot update yesterday. Does this resolve the issue? (Didn't reboot after the last 2 uboot updates).

Not had a chance to test myself yet. If you do test, please let us know results.

studioai, 2017-08-30:
@nphil 

20170824-9+deb8u1

is the version has the problem we discuss.

if you have a backup, just format your sd card with dietpi image and restore from backup.

rncwnd79, 2017-08-31:
Hmm. Downgrade not possible due to missing old versions:
```
root@odroid:~# apt-cache showpkg uboot
Package: uboot
Versions:
20170824-9+deb8u1 (/var/lib/apt/lists/fuzon.co.uk_meveric_dists_all_xu3_binary-armhf_Packages) (/var/lib/dpkg/status)
 Description Language:
                 File: /var/lib/apt/lists/fuzon.co.uk_meveric_dists_all_xu3_binary-armhf_Packages
                  MD5: 595d272760d8b921d997e57be519f219

Reverse Depends:
  linux-image-4.9.44+,uboot
  linux-image-4.9.43+,uboot
  linux-image-4.9.38+,uboot
  linux-image-4.9.37+,uboot
  linux-image-4.9.35+,uboot
  linux-image-4.9.34+,uboot
  linux-image-4.9.33+,uboot
  linux-image-4.9.30+,uboot
  linux-image-3.10.105+,uboot
Dependencies:
20170824-9+deb8u1 -
Provides:
20170824-9+deb8u1 -
Reverse Provides:
```

Fourdee, 2017-09-02:
Works fine for me, with current uboot package:
```
root@DietPi:~# uname -a
Linux DietPi 3.10.105+ #1 SMP PREEMPT Mon Aug 7 10:09:24 CEST 2017 armv7l GNU/Linux

root@DietPi:~# dpkg -l | grep uboot
ii  uboot                        20170824-9+deb8u1                armhf        U-Boot binaries for 5422
```

Can anyone else confirm?

nphil, 2017-09-02:
just did a uboot upgrade, same issue :(

rncwnd79, 2017-09-15:
Sorry for the late replay. I was on vacation.

Did the upgrade and a reboot. No problems!

Fourdee, 2017-09-16:
@rncwnd79 

Excellent, thanks for letting me know.

I'll mark this as resolved. Please reopen if required.

closed 2017-09-16 by Fourdee