MichaIng/DietPi #1127: Chromium Without desktop Fail start

qpens, 2017-09-01:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 155
- SBC Device oDroid C2
- Power supply used 5v 4A 
- SD card used Lexar
- Distro (EG: Jessie) | Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux
-LCD Waveshare 10.1 1280×800, IPS

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
autostart chromium

#### Actual behaviour:
Start fail 

[image]
[image]
[image]

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Invictaz, 2017-09-01:
Have the same issues on the x86_64 test build. Might be possible that this is GPU related? Or it displays the Chromium screen on the other ( side of) the display.

Fourdee, 2017-09-01:
@qpens 
Thanks for the report 👍 

I've personally only tested this with RPi. Does Chromium function correctly from Desktop launch?

qpens, 2017-09-01:
I didn't Install desktop I'm using only dietpi with octoprint just to run my 3d printer. I'm trying to set this thing so only chromium with octoprint will be showing up on start.

Fourdee, 2017-09-02:
@qpens 

Unable to replicate on a fresh install with 720P and 1080P. Chromium displays correctly (albeit, not full screen). May be limited to your monitor. Try changing the resolution in ```dietpi-config``` > display options.

Desktop is installed by default with Chromium. Try ```dietpi-autostart``` and changing to desktop.

qpens, 2017-09-02:
Waveshare instruction says put 
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1280 800 60 6 000
hdmi_drive=1

to config.txt
but that doesn't work I checked on UART terminal and its still loading 720p I try to change the resolution in dietpi-config  but there is only 720p and 1080p closest to my screen checked 1080p but it crashes the screen
I edit boot.ini and changed 720p to 800p but that doesn't work
How to force dietpi to use waveshare config?

qpens, 2017-09-02:
I reinstalled the OS and screen was working fine from v144 to V155 
[image]

after installing Octoprint screen got funny

[image]

Fourdee, 2017-09-03:
@qpens 

> Waveshare instruction says put
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 1280 800 60 6 000
hdmi_drive=1
in config.txt
but that doesn't work I checked on UART terminal and its still loading 720p

Those options are for RPi, wont have any effect on Odroid C2. Take a look at ```/DietPi/boot.ini```, there is a few HDMI options in there that may work.

qpens, 2017-09-03:
Before starting kernel

[image]

Desktop

[image]

changes in boot.ini didn't help

chromium still faill ;(

qpens, 2017-09-06:
Problem with screen Fixed

In config.txt
Block both framebuffer 
 
boot.ini 
 change resolution from me it was 1280x800p60hz

closed 2018-02-22 by Fourdee