MichaIng/DietPi #1128: XU4 Emby install failed

Snoevit, 2017-09-02:
**Required Information:**

**DietPi Version** | 154
**SBC Device** | Odroid XU4
**PSU** | Original 5v 4.0A  AC Adaptor
**SD card** | Sandisk ultra 16GB
**Distro** | Linux Embryoid 3.10.105+ #1 SMP PREEMPT Mon Aug 7 10:09:24 CEST 2017 armv7l GNU/Linux

**Additional Information (if applicable):**

**Software title?** Emby Server

Can this issue be replicated on a fresh installation of DietPi? This is a fresh install

**Expected behaviour:**

Complete Installation

**Actual behaviour:**

Error Code 100 Apt-get failed
(Apt string = emby server embymagick)

**Steps to reproduce:**

**Extra details:**

E: Unable to correct problems, you have held broken packages.
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

**contents of  #var/log/dietpisoftware_apt.log**:
E: Unable to correct problems, you have held broken packages.

Did run apt-get -f install & apt-get update with no errors, tried re-install with same outcome.

Fourdee, 2017-09-02:
@Snoevit 

Thanks for the report. 👍 

Confirmed here, we'll attempt to resolve.

Fourdee, 2017-09-02:
notes:
```
The following packages have unmet dependencies:
 embymagick : Depends: libjpeg8 (>= 8c) but it is not going to be installed
```

Fourdee, 2017-09-02:
Fixed for v155.

Workaround until its release is:
- Install ```libjpeg8``` prior to installing emby: https://github.com/Fourdee/DietPi/issues/1128#issuecomment-326743471

closed 2017-09-02 by Fourdee