MichaIng/DietPi #1130: DietPi-Config | DietPi - Filesystem benchmarks - RAM

k-plan, 2017-09-03:
## Creating a bug report/issue:
 Fail to execute:  DietPi - Filesystem benchmarks - RAM

[image]

```
(EE): Insufficient space to run RAM filesystem benchmark.
 - Available space: -10 MB
 - Space required: 10 MB

Press any key to continue.....
```

#### Required Information:
- DietPi Version |  v 155

- SBC Device | NanoPi Neo (armv7l)

```
───────────────────────────────────────
 DietPi     | 14:50 | Sun 03/09/17
 ───────────────────────────────────────
 V155       | NanoPi Neo (armv7l)
 ───────────────────────────────────────

```
- Power supply used : 5,1V 2,5A RPi-3 official PSU

- SD card used : Verbatim 8GB Class 10 Micro SDHC

- Distro (EG: Jessie) | Linux NanoPi-Neo 3.4.113-sun8i # 18 SMP PREEMPT Thu Jun 15 02:16:06 CEST 2017 armv7l GNU/Linux

#### Additional Information (if applicable):

- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Expected behaviour:

[image]

#### Actual behaviour:

```
 (EE): Insufficient space to run RAM filesystem benchmark.
 - Available space: -10 MB
 - Space required: 10 MB

Press any key to continue.....

```
#### Steps to reproduce:

`root@NanoPi-Neo:~# dietpi-config 12`

#### Extra details:

```
root@NanoPi-Neo:~# cat /proc/meminfo
MemTotal:         506768 kB
MemFree:          410400 kB
Buffers:            2292 kB
Cached:            25200 kB
SwapCached:            0 kB
Active:            52140 kB
Inactive:          19560 kB
Active(anon):      44212 kB
Inactive(anon):    11276 kB
Active(file):       7928 kB
Inactive(file):     8284 kB
Unevictable:           0 kB
Mlocked:               0 kB
HighTotal:             0 kB
HighFree:              0 kB
LowTotal:         506768 kB
LowFree:          410400 kB
SwapTotal:       1591292 kB
SwapFree:        1591292 kB
Dirty:               268 kB
Writeback:             0 kB
AnonPages:         44328 kB
Mapped:            16172 kB
Shmem:             11284 kB
Slab:              11812 kB
SReclaimable:       3960 kB
SUnreclaim:         7852 kB
KernelStack:         704 kB
PageTables:         1092 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     1844676 kB
Committed_AS:     150476 kB
VmallocTotal:     499712 kB
VmallocUsed:       17076 kB
VmallocChunk:     253036 kB
```
[image]

[image]

cu
k-plan

Fourdee, 2017-09-03:
@k-plan 

Thanks for the report 👍 

This is a limitation with the way we benchmark memory. We use the ```/tmp``` directory for the benchmark, which is mounted to RAM. It has a fixed size in most cases and we rely on a filesystem (of some kind) for all tests.

Command used to calc available space for RAM test:
```
df -m | grep -m1 '/tmp' | awk '{print $4}'
```

I'am unable to think of another method we could use, that does not rely on a file-system ```/tmp``` (mounted to RAM) :frowning_face:

k-plan, 2017-09-06:
@Fourdee 

Okay.

Sorry, but then I don't understand, why `dietpi-config 12 => RAM` works in previous versions of DietPi .
_< be at a loss >_

k-plan, 2017-12-07:
@Fourdee 

> Command used to calc available space for RAM test (returns MB):
>
>` df -m | grep -m1 '/tmp' | awk '{print $4}'`

Yes, sorry, and this is part of the problem. 
Where is no '`tmp`' (in RAM '`tmpfs`')
Only notice this on both of my Allwinner H3 ARMbian based devices.

```
root@NanoPi-Neo:~# df -m | grep -m1 '/tmp' | awk '{print $4}'
root@NanoPi-Neo:~#
```

```
root@NanoPi-Neo:~# df -h
Dateisystem    Größe Benutzt Verf. Verw% Eingehängt auf
/dev/mmcblk0p2  3,6G    1,3G  2,4G   35% /
udev             10M       0   10M    0% /dev
tmpfs            99M    4,5M   95M    5% /run
tmpfs           248M       0  248M    0% /dev/shm
tmpfs           5,0M       0  5,0M    0% /run/lock
tmpfs           248M       0  248M    0% /sys/fs/cgroup
tmpfs            10M    1,4M  8,7M   14% /DietPi
tmpfs            20M     72K   20M    1% /var/log
/dev/mmcblk0p1   64M     18M   47M   27% /boot
tmpfs            50M       0   50M    0% /run/user/999
```

**Cross-check:**

**RPi3** (and RPi Zero W)

```
root@RPi3:~# df -m | grep -m1 '/tmp' | awk '{print $4}'
470

root@RPi3:~# df -h
Dateisystem                                 Größe Benutzt Verf. Verw% Eingehängt auf
/dev/root                                     15G    4,6G  9,7G   32% /
devtmpfs                                     466M       0  466M    0% /dev
tmpfs                                        470M       0  470M    0% /dev/shm
tmpfs                                        470M    6,4M  464M    2% /run
tmpfs                                        5,0M    4,0K  5,0M    1% /run/lock
tmpfs                                        470M       0  470M    0% /sys/fs/cgroup
tmpfs                                         10M    1,5M  8,6M   15% /DietPi
tmpfs                                         20M     32K   20M    1% /var/log
tmpfs                                        470M     32K  470M    1% /tmp
/dev/mmcblk0p1                                63M     23M   41M   36% /boot
tmpfs                                         94M       0   94M    0% /run/user/0
```
and

**Odroid C2** (and Odroid C1)

```
root@Odroid-C2:~# df -m | grep -m1 '/tmp' | awk '{print $4}'
860

root@Odroid-C2:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   15G  1.6G   13G  11% /
udev             10M     0   10M   0% /dev
tmpfs           344M   15M  330M   5% /run
tmpfs           860M     0  860M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           860M     0  860M   0% /sys/fs/cgroup
tmpfs            10M  1.2M  8.9M  12% /DietPi
tmpfs            20M  3.8M   17M  19% /var/log
tmpfs           860M     0  860M   0% /tmp
/dev/mmcblk0p1  128M   46M   82M  36% /boot
root@Odroid-C2:~#
```

But better we close this old issue.

k-plan, 2017-12-08:
Please reopen, if needed or someone will work on it. Thanks.

Fourdee, 2017-12-08:
@k-plan 

> Yes, sorry, and this is part of the problem.
There is no 'tmp' (in RAM 'tmpfs')
Notice this first on both of my Allwinner H3 ARMbian based devices.

Yep, we disable ```/tmp``` as RAM for <= 512MB devices: https://github.com/Fourdee/DietPi/issues/1027

Fourdee, 2017-12-08:
I'll add a check in for ```/tmp``` as ```tmpfs``` before allowing to run RAM bench.

k-plan, 2017-12-08:
@Fourdee 

👍  thanks for the link! 😃 

😢  who the hell needs Microsoft's .NET Framework on his linux SBC ?
I'm happy to to be rid of this on my desktop  !   👍  😄 

Will be nice you disable this for the affected devices. 👏

Fourdee, 2017-12-08:
[image]

k-plan, 2017-12-08:
Wow, that was now a super quick fix after 4 month!

Many thanks! 💯  👍

Fourdee, 2017-12-08:
Confirmed on RPi Zero.

Completed.

closed 2017-12-08 by Fourdee