MichaIng/DietPi #1131: Issues downloading/installing via dietpi-software

samteezy, 2017-09-03:
## Creating a bug report/issue:
Hi, I'm new to DietPi as of today - formerly Raspian Lite user. I think I'm seeing an issue in which the use of pi-hole conflicts with dietpi's handling of services out of the box.

Ever since I installed and activated pi-hole I cannot install packages via dietpi-software because, from what I can tell, it can't resolve DNS queries. I think this is because dietpi disables dnsmasq when it starts the install process, and my RPi is the primary DNS server on my LAN. For example, I've been trying to install phpsysinfo, and I get the "unable to install" because the URL "is offline and/or unreachable" error. Similarly, the scripted repo update errors during this process too.

However, after the process completes and the PI reboots, I can wget the package with no issue. I've tried this with multiple packages.

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 155
- SBC Device (EG: RPi 3)? RPi 3
- Power supply used (EG: 5v 1A RAVpower)? 2.5A
- SD card used (EG: Sandisk ultra)? SanDisk Ultra 32GB
- Distro (EG: Jessie) | ```uname -a```? jessie

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
Install should complete.

#### Actual behaviour:
URL doesn't resolve and package can't complete.

#### Steps to reproduce:
1. Install pi-hole. Configure as primary DNS on LAN.
2. Attempt to install package via dietpi-software.

#### Did you submit a ```dietpi-bugreport```?
74bfe7cd-1c38-4d8f-9809-d1a8a1c50545-0

#### Extra details:
None at this time.

k-plan, 2017-09-03:
hi,
can you please show us the output of:

`cat /etc/pihole/install.log`

and 

`cat /etc/pihole/setupVars.conf`

samteezy, 2017-09-03:
Network settings (I'm not using wifi, btw). As you can see my local device IP is also my DNS server.
```cat /etc/network/interfaces
#/etc/network/interfaces
#Please use DietPi-Config to modify network settings.

# Local
auto lo
iface lo inet loopback

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.50.24
netmask 255.255.255.0
gateway 192.168.50.1
dns-nameservers 192.168.50.24

# Wifi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
wireless-essid Not_Set
wireless-key Not_Set
wireless-mode Managed
wireless-power off
wpa-ssid Not_Set
wpa-psk Not_Set
#dns-nameservers 8.8.8.8 8.8.4.4
```

Pihole:

```
cat /etc/pihole/setupVars.conf
WEBPASSWORD=[redacted]
DNSMASQ_LISTENING=single
PIHOLE_DNS_3=2607:5300:60:815b:db25::1
PIHOLE_DNS_4=2601:282:8201:5d00::feed
DNS_FQDN_REQUIRED=true
DNS_BOGUS_PRIV=true
DNSSEC=false
PIHOLE_INTERFACE=eth0
IPV4_ADDRESS=192.168.50.24/24
IPV6_ADDRESS=
PIHOLE_DNS_1=192.95.54.1
PIHOLE_DNS_2=96.90.175.167
QUERY_LOGGING=true
INSTALL_WEB=true
API_EXCLUDE_DOMAINS=
API_EXCLUDE_CLIENTS=
API_QUERY_LOG_SHOW=blockedonly
API_PRIVACY_MODE=false

```

Nothing standing out in the last two logs you requested, no errors during install. I've also run pihole -r to be sure it wasn't a setup issue.

k-plan, 2017-09-03:
> As you can see my local device IP is also my DNS server.

> `dns-nameservers 192.168.50.24`

... and this is the mistake!

Use what ever you like, 8.8.8.8, 8.8.4.4, or DNS Server of your ISP or DNS provider by your default gateway (router for your ISP)  (192.168.50.1)

samteezy, 2017-09-03:
I'm aware of that. In and of itself isn't a problem though... it's how dietpi decides to handle services. If it didn't stop dnsmasq and other services when running an install of an application, then this wouldn't be a concern.

192.168.50.1 is my router. Guess what the primary DNS is on the router... it's the RPi (192.168.50.24). This is exactly how pi-hole is supposed to be used, and has worked perfectly fine for the better part of a year.

Why exactly does DietPi need to stop services when installing programs via dietpi-software? I could easily install many of these myself via apt-get, but I'm trying to leverage all the features of DietPi since the contributors have clearly put a great deal of time and effort into it.

samteezy, 2017-09-04:
Okay, I think I might know why the changes in my last edit didn't work. I don't have it in front of me but I know that after applying the network setting changes, I noticed an error message in regards to eth0 reconnecting.

After rebooting the device, I don't see that message when changing network settings and everything's working the way I want. I can have the RPi IP set as the primary DNS, but when I disable the services, it can still resolve domains.

Thanks all, and especially Fourdee, as I'm becoming impressed by these applications you've built to make working in the terminal that much easier.

closed 2017-09-16 by samteezy