MichaIng/DietPi #1132: wifi configuration fails when key contains = character

gretel, 2017-09-04:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 1.55
- SBC Device | Rpi Zero W
- Power supply used | industrial
- SD card used | samsung 16gb
- Distro | jessie

#### Expected behaviour:
wifi should connect
#### Actual behaviour:
various errors messasges related to `wlan0` appearing - wlan not functional

#### Steps to reproduce:
set a `Wifi_KEY` containing `"` in `dietpi.txt`

k-plan, 2017-09-04:
> In the first place, the choice of a secure WPA network key (also known as a passphrase or a pre-shared key) should be selected during the PSK process. This should use the maximum key length of 63 characters.  ...
> However, special characters in the password may cause problems with some operating systems, especially with Apple iOS. Particular caution is required with special characters from the international language area (eg ü, ö, ä ,ß or also § ). Same for keys contains characters which are not permitted '~!#$%^&@()=[]}{\|;:'<>/ °$ € ł µ ² ³ ` ? and so on.  Depending on the operating system ( Microsoft Windows , Mac OS X , Unix ), they are encoded differently and are not compatible with each other.

gretel, 2017-09-05:
@k-plan i was not referring to any wifi-related in general, and i am quite aware of the limitations, thanks for pointing that out.
`"` is a valid character. the documentation nor inline comments in `dietpi.txt` many any restrictions in regards to not being allowed to use the `=` character.
i consider this as a bug because it would require me to change the access point configuration and all nodes connected to have them join a '`=`' free setup - which is quite a hassle, honestly.
the exception is dietpi - i guess there is no or proper string escape in place. the tools like `sed` etc. work all fine if used properly.

hippytyre, 2018-01-10:
I think I'm having the same problem. The WIFI SSID I'm trying to connect my Pi Zero W to is -=word=- I've tried to put it into quotation marks but this doesn't seem to help. Has this issue been looked at or does anyone have a workaround? Changing the SSID is not an option I'm afraid.

Fourdee, 2018-01-11:
@hippytyre 

Hi,

Try escaping each ```\=``` entry, when entering the SSID:
```-\=word\=-```

Fourdee, 2018-02-22:
Outdated.

closed 2018-02-22 by Fourdee