MichaIng/DietPi #1133: setting prefer_ipversion=ipv4 has no effect - ipv6 is preferred

gretel, 2017-09-04:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 1.55
- SBC Device | Rpi Zero W
- Power supply used | industrial
- SD card used | samsung 16gb
- Distro | jessie

#### Expected behaviour:
setting should be respected - `apt` should not connect via ipv6
#### Actual behaviour:
ipv6 is preferred - regardless of the setting

#### Steps to reproduce:
set `prefer_ipversion=ipv4` on a ipv6 enabled system and watch `apt` connecting via ipv6

MichaIng, 2017-09-05:
Hmm so there are szenarios where you need IPv4 for some connections while IPv6 stays in place?
In dietpi-config you could completely disable the IPv6 kernel module. In the past this also added net.ipv6.conf.all.disable_ipv6=1 but we just removed that as it produced errors in log if the module was anyway not loaded.
I will have a look later if this somehow produces the error you face when IPv6 module stays enabled.

MichaIng, 2017-09-05:
"for now only apt will not connect via ipv6." Just read about it: https://unix.stackexchange.com/questions/9940/convince-apt-get-not-to-use-ipv6-method

My impression is, that if you generally want to use IPv6 you should figure out why apt fails.
By using net.ipv6.conf.all.disable_ipv6=1 you anyway disable IPv6 for everything and in this case it makes mor sense (for me) to completely disable the kernel module, which can be done via dietpi-config.

gretel, 2017-09-05:
@MichaIng my native ipv6 connection is fine. but i consider it as a bug because it effectively prevents the installation from finishing, as for some reason unknown to me the raspbian redirector does not respond in time. considering this an alternative may be to configure a connection timeout for `apt` and leave things as they are - which would still be confusing if users do expect things to work as configurable in `dietpi.txt`.

gretel, 2017-09-05:
https://github.com/Fourdee/DietPi/issues/1119 looks related to me.

MichaIng, 2017-09-05:
Ah it's the default Raspbian redirector that fails on IPv6? I don't have IPv6 configured, so can't try to replicate it, I guess fourdee can.

MichaIng, 2017-11-10:
Found another issue related to this:
Even that ipv6 module is blacklisted (which is what dietpi-config does on disabling IPv6), wget and apt sometimes (depending on repo) try to connect to IPv6 addresses. This was the case for me with owncloud installation via their own repo. I needed to adjust setting in /etc/.wgetrc to get repo key via wget and switching repo address from http to https somehow made apt choose IPv4.

Adding the sysctl setting mentioned in OP produces error if module is blacklisted but if it is the only way to force apt (and wget) to prefer IPv4, we should set it anyway.

Will investigate that further...

MichaIng, 2018-01-26:
Close as this should have been solved by recent rework. Reopen on demand.

closed 2018-01-26 by MichaIng