MichaIng/DietPi #1134: password set in dietpi.txt (AUTO_Global_Password) does get reset to default on upgrade of dietpi itself

gretel, 2017-09-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 1.55
- SBC Device | Rpi Zero W
- Power supply used | industrial
- SD card used | samsung 16gb
- Distro | jessie

#### Expected behaviour:
password should keep unchanged as configured in `dietpi.txt`
#### Actual behaviour:
password gets reset on `dietpi-update`

#### Steps to reproduce:
install 1.50 from image with `AUTO_Global_Password` set and changed from the default in `dietpi.txt`. when the installation finished run `dietpi-update`, `reboot` and try to log in with the password expected.

Fourdee, 2017-09-05:
@gretel 

Thanks for the report 👍 

We'll investigate.

mpember, 2017-09-09:
I found this to be the case with a similar build (from a v150 image).  Since it was a clean install and I had never used this method previously, I wasn't sure if it was a feature or a bug.  I can confirm that the root password remained as set in dietpi.txt.  But when I first went to log in as "dietpi", the password was the normal default of "dietpi".

Fourdee, 2017-09-09:
@mpember 

>  I can confirm that the root password remained as set in dietpi.txt

Excellent.

> But when I first went to log in as "dietpi", the password was the normal default of "dietpi".

Yep, currently only the root password is set.

I think for now, we could change both "root" and "dietpi" user passwords as set in dietpi.txt.
But in the future, we will only set "dietpi" user and disable "root" at some point: https://github.com/Fourdee/DietPi/issues/896

Fourdee, 2017-10-11:
I'll mark this completed: https://github.com/Fourdee/DietPi/commit/435c72ddc2378b454cca133c815e464c1ef292c4

@gretel 
I believe the following chars may of been used.
```
WARN: Passwords with the any of the following characters are not supported: \"$
```

If not, please can you paste the password used so we can test and reopen ticket?

closed 2017-10-11 by Fourdee