MichaIng/DietPi #1135: Perl problem / unable to install Pi-Hole (unable to stat `./usr/share/perl/5.20.2/CPAN/API/HOWTO.pod)

jwgn, 2017-09-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version  **155**
- SBC Device **RPi 1**
- Power supply used **5v 2.5A**
- SD card used **Verbatim 8 GB**
- Distro (EG: Jessie) **Linux DietPi 4.9.40+ #1 Sat Aug 5 10:51:04 UTC 2017 armv6l GNU/Linux**

#### Additional Information (if applicable):
- Software title? **Pi-Hole**
- Can this issue be replicated on a fresh installation of DietPi? **Yes**

#### Expected behaviour:
DietPi should install Pi-Hole.

#### Actual behaviour:
 [Failed] Apt simulation | An issue has occured

 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues

#### Steps to reproduce:
Connect to the Pi via SSH and continue with the installation

#### Did you submit a ```dietpi-bugreport```?
5c18265b-560f-412b-a9bc-6ac5afe4c748-0

#### Extra details:

apt mirror has been changed - no effect, same error
apt cache has been refreshed

'apt-get install -f' returns:

root@DietPi:~# apt-get install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following package was automatically installed and is no longer required:
  libpci3
Use 'apt-get autoremove' to remove it.
The following extra packages will be installed:
  perl-modules
Suggested packages:
  libb-lint-perl libcpanplus-dist-build-perl libcpanplus-perl libfile-checktree-perl libobject-accessor-perl
Recommended packages:
  libpod-latex-perl libtext-soundex-perl
The following packages will be upgraded:
  perl-modules
1 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
1 not fully installed or removed.
Need to get 2,547 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 ftp://ftp.fu-berlin.de/linux/raspbian/raspbian/ jessie/main perl-modules all 5.20.2-3+deb8u8 [2,547 kB]
Fetched 2,547 kB in 1s (1,581 kB/s)
(Reading database ... 19168 files and directories currently installed.)
Preparing to unpack .../perl-modules_5.20.2-3+deb8u8_all.deb ...
Unpacking perl-modules (5.20.2-3+deb8u8) over (5.20.2-3+deb8u6) ...
dpkg: error processing archive /var/cache/apt/archives/perl-modules_5.20.2-3+deb8u8_all.deb (--unpack):
 unable to stat `./usr/share/perl/5.20.2/CPAN/API/HOWTO.pod' (which I was about to install): Structure needs cleaning
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/perl-modules_5.20.2-3+deb8u8_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Fourdee, 2017-09-09:
@jwgn 

Thanks for the report 👍 

We were unable to replicate this in our tests, may be limited to your current setup:
```
Setting up perl-modules (5.20.2-3+deb8u8) ...
Setting up perl (5.20.2-3+deb8u8) ...
```

Lets try removing the offending files and reinstall perl:
```
rm /usr/share/perl/5.20.2/CPAN/API/HOWTO.pod
apt-get clean
rm /var/cache/apt/archives/perl-modules_5.20.2-3+deb8u8_all.deb
apt-get update
apt-get install perl perl-modules --reinstall
```

jwgn, 2017-09-10:
Hm even rm did not work (Structure needs cleaning). It seems my SD card is just corrupted, e2fsck fixed a few errors, now installation resumes. I have ordered a new SD card. Thanks for your help and sorry for bothering with that.

closed 2017-09-10 by jwgn