MichaIng/DietPi #1139: 64bit possibility on RPi3?

mok-liee, 2017-09-08:
Hey @Fourdee,

is it possible to run DietPi on RPi3 under 64-bit with his armv8 CPU?

Greetings mok

mpember, 2017-09-08:
I think you the key roadblock is upstream development.  Until Raspbian gets a well-supported 64-bit release, it is a bit of a stretch (no pun intended) to expect the smaller distros to go it alone.

Fourdee, 2017-09-09:
> Until Raspbian gets a well-supported 64-bit release, it is a bit of a stretch (no pun intended) to expect the smaller distros to go it alone.

Could not of put it better myself 👍 , exactly correct. We simply don't have the resources to fully support and assist with additional "in-development/testing" images at this time.

When Raspbian officially releases ARM64 image for RPi, we'll follow.

mok-liee, 2017-09-10:
thank you very much... i will wait :)

closed 2017-09-10 by mok-liee