MichaIng/DietPi #1140: MPD dependency links lost

mpember, 2017-09-08:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
155
- SBC Device (EG: RPi 3)?
RPi 3
- Power supply used (EG: 5v 1A RAVpower)?
Official RPi PSU
- SD card used (EG: Sandisk ultra)?
Sandisk Ultra
- Distro (EG: Jessie) | ```uname -a```?
Jessie

#### Additional Information (if applicable):
- Software title?
MPD
- Can this issue be replicated on a fresh installation of DietPi?
This occurred on a clean install

#### Expected behaviour:
MPD and related dependencies should be installed and flagged as still required

#### Actual behaviour:
Upon performing any further apt-get requests, the user is prompted to perform an 'autoremove' request to remove packages identified as no longer being required.  These include all dependency packages installed by the MPD installation.

#### Steps to reproduce:
1. Install MPD using the dietpi-software method.
2. Install any other program via call to apt-get.  In my case, I has installed a third-party repo to install upmpdcli

#### Did you submit a ```dietpi-bugreport```?

No

#### Extra details:

Fourdee, 2017-09-09:
@mpember 

Thanks for the report 👍 

I believe this is due to:
- 1st we install MPD via the debian repo (to pull in the shared packages)
- Then we install our updated version of MPD via dpkg

This does not apply to our stretch, same version of MPD installed.

Notes:
- Remove our MPD compiled binary, simply install MPD via repo as we now offer Stretch images for our major boards.

mpember, 2017-09-09:
Thanks.  I had already got it working (preemptively using the steps you have outlined).  I figured I'd post a detailed report to avoid someone posting a less-detailed report.

Fourdee, 2017-09-09:
@mpember 

> I figured I'd post a detailed report to avoid someone posting a less-detailed report.

Much appreciated 👍

Fourdee, 2017-09-09:
Tested:
- 🈯️ Jessie
- 🈯️ Stretch
- 🈯️ ARM64

Fixed for v156.

closed 2017-09-09 by Fourdee