MichaIng/DietPi #1143: Image request: NanoPI Neo Plus2

dpskvn, 2017-09-11:
It seems the NanoPi Neo2 image could be tweaked to be compatible: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Plus2

Major differences are included 8GB eMMC, addition of WiFi/Bluetooth, additional USB port and RAM increased to 1GB.

I successfully booted the NanoPI Neo2 image, although only half of the RAM is available and WiFi is not working.

I don't necessarily feel comfortable poking around myself, but if you need me to run commands and test stuff, I'd be happy to.

I did notice something while running `lshw`, though:

```
    description: Computer
    product: Xunlong Orange Pi PC 2
    serial: 8280000114ce2103
    width: 64 bits
  *-core
       description: Motherboard
       physical id: 0
       capabilities: xunlong_orangepi-pc2 allwinner_sun50i-h5
     *-cpu:0
          description: CPU
          product: cpu
          physical id: 0
          bus info: cpu@0
     *-cpu:1 DISABLED
          description: CPU
          product: cpu
          physical id: 1
          bus info: cpu@1
     *-cpu:2 DISABLED
          description: CPU
          product: cpu
          physical id: 2
          bus info: cpu@2
     *-cpu:3 DISABLED
          description: CPU
          product: cpu
          physical id: 3
          bus info: cpu@3
     *-memory
          description: System memory
          physical id: 4
          size: 483MiB
```

So I tried the OrangePI PC 2 image, but still no WiFi, eMMC or extra RAM.

exside, 2017-12-07:
Anything that would help making this real? Sending you one @Fourdee ?

Fourdee, 2018-07-09:
Marking as closed.

If we receive, or, are able to obtain the device, we'll reopen for image implementation.

closed 2018-07-09 by Fourdee

aliosa27, 2018-07-16:
Reopening. I did the work for this device a while ago, and have been using it ever since, ill start clean and have a beta in a few days

aliosa27, 2018-07-16:
I guess I cant reopen it, but ill geterdone