MichaIng/DietPi #1145: Native PC - Home Assistant installation fails

hiscorebob, 2017-09-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version |  155
- SBC Device |  Native PC
- Power supply used | Built-in
- SD card used | S-ATA Drive
- Distro (EG: Jessie) | 4.9.0-3-amd64 #1 SMP Debian 4.9.30-2+deb9u3 (2017-08-06) x86_64 GNU/Linux

#### Additional Information (if applicable):
- Software title? | Home Assistant
- Can this issue be replicated on a fresh installation of DietPi? | Yes

#### Expected behaviour:
Home Assistant should be installed

#### Actual behaviour:
Installation stops with following error message :
E: Package 'libmysqlclient-dev' has no installation candidate
All Services remain stopped
#### Steps to reproduce:
Select "Home Assistant" for Installation
Start Installation
#### Did you submit a ```dietpi-bugreport```?
6991a32f-cdb2-48ba-bae5-cfd82a846cc5-0

#### Extra details:
Tested other repositories (Global/US/UK)
Refreshed package info (apt-get update)

Fourdee, 2017-09-13:
@hiscorebob 

Thanks for the report 👍 

Package is unavailable in Stretch:
https://packages.debian.org/jessie/libmysqlclient-dev

As MariaDB is now the default SQL install on Stretch, this would need to be installed in its place:
https://packages.debian.org/stretch/libmariadbclient-dev

Fourdee, 2017-09-13:
@sytone 

Are you available to make the package changes based on ```$DISTRO``` in ```dietpi-software```? (3=jessie,4=stretch)
https://github.com/Fourdee/DietPi/issues/1145#issuecomment-329201055

sytone, 2017-09-17:
Will libmariadbclient-dev still work for python connections to MySQL databases. This was added as this is the common one used in HA installations to make configuration easier.

Fourdee, 2017-09-18:
@sytone 
I cant see it being an issue, MariaDB==MySQL but we'd need to test it.

sytone, 2017-09-23:
ok, ill try and get to it in the next few days. Is there a VM image for stretch around? If not what process are you using to make them. I'm taking the virtualbox images and converting them to hyper-v to make life simple for me.

Fourdee, 2017-09-23:
@sytone 

>  Is there a VM image for stretch around?

Jessie currently.

You can upgrade:
- Fresh install of VM image and DietPi setup completed with no changes
Copy and paste all
```
DISTRO='stretch'
cat << _EOF_ > /etc/apt/sources.list
deb http://ftp.debian.org/debian/ $DISTRO main contrib non-free
deb http://ftp.debian.org/debian/ $DISTRO-updates main contrib non-free
deb http://security.debian.org $DISTRO/updates main contrib non-free
deb http://ftp.debian.org/debian/ $DISTRO-backports main contrib non-free
_EOF_
```

Then
```
apt-get update
apt-get dist-upgrade
```

Should be fine to use default options if any whiptail menus appear.

sytone, 2017-10-20:
I have a update for ha in progress to deal with python. I’ll be adding this next.

Fourdee, 2017-10-26:
Merged into testing branch : https://github.com/Fourdee/DietPi/pull/1191

Ready for testing:
https://github.com/Fourdee/DietPi/blob/master/TESTING-BRANCH.md#steps-to-use-the-dietpi-testing-branch

hiscorebob, 2017-11-08:
I have tested the "testing" branch, and can confirm that the Problem is solved. Home assistant installs and runs as expected.

closed 2017-11-08 by hiscorebob