MichaIng/DietPi #1146: [APT][Optimization] Remove/Ignore recommendations | apt-mark auto

MichaIng, 2017-09-13:
As I am a perfectionist, I always try to optimize my systems as much as I can, trying and tinkering around. For reducing the amount of unnecessary installed apt packages, you could remove all automatically installed "recommended" packages, so packages that were just recommended but no "dependency" of manually installed packages. DietPi already uses this when installing it's needed packages for system preparation: https://github.com/Fourdee/DietPi/search?utf8=✓&q=recommends&type=

There is a way to force this setting for all `apt(-get) install` calls by adding the following to some /etc/apt/apt.conf.d/ file:
```
APT::Install-Recommends "false";
APT::Install-Suggests "false"; //This one disables auto installation of "suggestions", but it's actually default in debian.
```
By adding the following again, you can further force `apt(-get) autoremove` to also remove all recommendations (and suggestions):
```
APT::AutoRemove::RecommendsImportant "false";
APT::AutoRemove::SuggestsImportant "false";
```
Afterwards `apt update && apt autoremove --purge` will remove all automatically by apt installed packages, that are not indeed necessary to run the manually installed ones. This found for me quite a lot, even library packages, at least after the fresh DietPi RPi Stretch Beta setup.
____________
For this to work effectively, the system needs to know, which packages are indeed manually installed/explicitly wanted by the user (or DietPi to run it's own scripts). I found a huge bunch of packages marked as "manually" installed, including many lib* packages, after fresh DietPi setup, which can be found by `apt-mark showmanual`. This comes from the Raspbian image, which is by far not intended to be lightweight, so I don't know if it's the same for Debian images, but I guess. All these packages will thus never be removed by `apt(-get) autoremove`. I manually marked many packages as `apt-mark auto` to enable `autoremove` to remove those together with all of their recommendations, which were again quite a lot.
<details>
<summary>my output of "apt-mark showmanual"</summary>

```
apache2 //webserver stack
apt
apt-transport-https
apt-utils
ca-certificates
cron
curl
debconf-utils
dphys-swapfile
dropbear
ethtool
fake-hwclock
fbset
findutils
hdparm
htop
insserv
iptables //firewall
isc-dhcp-client
libapache2-mod-php //webserver stack
libpam-systemd
libraspberrypi-bin //Raspberry Pi specific
locales
logrotate //logging type 2
mariadb-server //webserver stack
mount
mysqltuner //mysql/mariadb analyzation
nano
net-tools
ntp
parted
php //webserver stack
php-apcu //Nextcloud
php-curl //webserver stack
php-gd //webserver stack
php-intl //Nextcloud
php-mbstring //webserver stack
php-mysql //webserver stack
php-redis //Redis
php-xml //webserver stack
php-zip //webserver stack
psmisc
python-certbot-apache //letsencrypt, I prefer the default apt package here
raspberrypi-sys-mods //Nextcloud
raspbian-archive-keyring //Nextcloud
raspi-copies-and-fills //Nextcloud
redis-server //Redis
resolvconf
rsync
rsyslog //logging type 2
screen //SSH screen sessions, love this one!
sudo
tzdata
unzip
usbutils
wget
whiptail
wput
zip
```
</details>

Full amount of installed packages is 343.

Many packages are due to my Nextcloud instance, others are RPi specific. All packages related to WiFi and bluetooth, as well as some archive and file system packages I personally don't need are removed. Therefore some packages, installed by dietpi system preparation script are missing. But basically the list in that script should match the list of `apt-mark showmanual` on fresh installations, in my opinion.

E.g. the system preparation script could first mark ALL packages as auto. The ones installed by the script will anyway automatically marked as manual by `apt(-get) install` afterwards. That way we would get a reduced image size and enable the system and user to identify/remove obsolete/unused packages easier.
_______________
Last but not least, I found some obsolete packages by time, that were not found in the repositories anymore. The gui package manager synaptic e.g. automatically lists them, aptitude allows to remove them by `aptitude purge ~o` but apt does not provide a way to show or autoremove those.
To identify all obsolete packages, we can use `dpkg -l | grep -v 'ii '`. The first two letters of the line show the status of the package: https://askubuntu.com/questions/18804/what-do-the-various-dpkg-flags-like-ii-rc-mean/18807#18807
`ui` do show installed packages, that are not part of the repo(s) anymore (observed this here and there after a while also with official Raspbian, due to package updates/dependency changes), other letters show e.g. remaining configuration files `rc`, if `--purge` was forgotten or something.
`dpkg --purge` could then remove identified packages. But if considered, this should be done in a very early step in system preparation script, because here and there are packages (e.g. apt-transport-https) installed by other sources than repo.

Removing obsolete packages could be also included into dietpi cleaner.
_____________________
I did all of this to clean up my system as much as I can and ran it for a while now without any problems related to this so far. Problems only appeared because I also tried to remove some packages that are also part of dependency list in system preparation script 😆, perfectionist, but it's nice to finally know, WHY they are dependencies for DietPi 👍 .

If this is wanted for all DietPi images, testing should be indeed done for all systems.

Fourdee, 2017-09-13:
@MichaIng 

> APT::Install-Recommends "false";
APT::Install-Suggests "false"; //This one disables auto installation of "suggestions", but it's actually default in debian.

Love the idea, I'am all for minimal, but this may opens a can of worms. Recommended packages in Debian are very "hit and miss".
Some cases, library packages (such as codecs) will be pulled in as recommended packages, without those, the majority of features will not be available, eg: lack mp3 playback support etc.

Although saying that, following devices (quick check) have recommended and suggested disabled by default:
- Sparky Stretch
- XU4 Jessie
```
root@DietPi:~# cat /etc/apt/apt.conf.d/71-no-recommends
APT::Install-Recommends "0";
APT::Install-Suggests "0";
```

> ui do show installed packages, that are not part of the repo(s) anymore (observed this here and there after a while also with official Raspbian, due to package updates/dependency changes), other letters show e.g. remaining configuration files rc
Removing obsolete packages could be also included into dietpi cleaner.

Sounds good 👍 
We should also check for this during system prep.sh and/or finalise scripts.

MichaIng, 2017-09-14:
> Some cases, library packages (such as codecs) will be pulled in as recommended packages, without those, the majority of features will not be available

Ah yes I remember that e.g. fail2ban just recommends iptables but indeed absolutely needs it to work. Maybe another firewall would also work why it's not set as dependency... As my system at the moment doesn't  run much more than Nextcloud, it works very well without any recommndation.
To still benefit from this, all recommendations could be removed during system prep script and afterwards they could be enabled again for further installations (in case they're not disabled by default).
For dietpi-software the single packages could be tested without recommendations by time and then in case recs could be excluded via apt command option.

I think the apt-mark thing, to at the beginning mark all packages as automatically installed, is the other important basis to reduce installed packages/image size.

closed 2017-12-27 by MichaIng