MichaIng/DietPi #1147: Minio Installer -- A Few Questions

techdabbler, 2017-09-13:
Hi,

I'm working on adding Minio to the DietPi-Software script. I've got it mostly working right now just a few tweaks to do with configuration options.  I have 2 questions I'd appreciate some direction on.

1) How/where can I put true uninstall steps? I saw a section at the bottom of the script that appeared to be the uninstall script section but I wanted to check.'

2) Once done do how do I get the updates to you? Do I need to go through git? Or can I just send the file somewhere?

Thanks for a great distro! It is definitively my new goto.

Fourdee, 2017-09-14:
@techdabbler 

Hi,

1. Create a new entry anywhere here for uninstall code:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L11662-L11666

2. Ideally, should be sent as a pull request on git :) Get full credit for your contributions :)

techdabbler, 2017-09-19:
Thanks. I got the uninstall all setup and I think I've figured out git enough to send it that way.  

I ran into one 'issue' that I found a way around but wanted to check to see if this dietpi longterm a good idea.

In short I make a folder for the server data in /mnt/dietpi_userdata/minio-data  where I need a user (minio-user) to have have rwx permissions.  Originally in the KISS method I tried to chown the folder with minio-user. When manually run it works. When run from the dietpi_software script it 'fails'. I believe what is actually happening is somewhere in the script it is setting the ownership back to dietpi.  

If I change Minio to run as dietpi user it works fine.  However, due to wanting to download the startup script and not having to edit it on the fly I'd much rather run as minio-user. I'd also rather NOT give all 'Other' rwx access if I can help it for security reasons.

So I used 
`setfacl -R -m u:minio-user:rwx /mnt/dietpi_userdata/minio-data/`

It works great while on the SD card but my concern is making sure it follows if users move /mnt/dietpi_userdata . If the drive is ext4 I assume it will follow it correctly but I wasn't sure if dietpi is forcing any particular level of fs to ensure this works.  

I'm also wide open to a better way to set the permissions so they are more portable.

Appreciate any ideas/thoughts.

Fourdee, 2017-09-20:
@techdabbler 

> If the drive is ext4 I assume it will follow it correctly but I wasn't sure if dietpi is forcing any particular level of fs to ensure this works.

When formatting drives in ```dietpi-drive_manager```, and moving using data, we prompt user if filesystem does not support fs permissions and the issues it may cause.

Add the chown entry here:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L7679

Gets applied at the end of ```dietpi-software``` installation.

closed 2017-10-19 by techdabbler