MichaIng/DietPi #1149: PiHole and Wifi Hotspot

denis-ev, 2017-09-14:
## Creating a bug report/issue:
When PiHole is installed and I try to install the Wifi Hotspot the service isc-dhcp-server fails to start.

#### Required Information:
- DietPi Version 155
- SBC Device RPi 3
- Power supply used 5v 1A
- Distro Linux hassio 4.9.48-v7+ #1034 SMP Fri Sep 8 14:31:14 BST 2017 armv7l GNU/Linux

#### Actual behaviour:
isc-dhcp-server fails to start

#### Steps to reproduce:
install pihole and the wifi hostspot

#### Extra details:
I like to set up my pi as a Hotspot and use the DHCP from PiHole like my LAN network.
Maybe you could create a packages, where pihole and hostspot are always installed together, to fix some config files.

enolp, 2017-09-26:
I reported the same problem months ago but the solution @k-plan gave me it didn't work for me. My bug report https://github.com/Fourdee/DietPi/issues/992

Fourdee, 2018-04-05:
Marking as closed due to:
- Inactivity, lack of interest from our user base and/or dev team.

closed 2018-04-05 by Fourdee