MichaIng/DietPi #1150: Error with Emby Server

damien92220, 2017-09-16:
Hi all,
error when i want install Emby Server

## Creating a bug report/issue:

#### Required Information:
- DietPi Version: 155
- SBC Device: Asus Tinker Board
- Power supply used: 5v 3A

#### Additional Information (if applicable):
- Emby Server
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Actual behaviour:
Apt-get has failed:
- Apt string = emby-server embymagick
- Error code = 100
- Software Title = Emby Server: web interface media streaming server

#### Steps to reproduce:
With new clean install

#### Did you submit a ```dietpi-bugreport```?
dietpi-software_apt.log = E: Unable to correct problems, you have held broken packages.

Fourdee, 2017-09-18:
@damien92220 

Thanks for the report 👍 

We'll investigate.

Fourdee, 2017-09-18:
```
package.deb         100%[===================>] 118.69K   347KB/s    in 0.3s

2017-09-18 18:14:32 (347 KB/s) - ‘package.deb’ saved [121540/121540]

Selecting previously unselected package libjpeg8:armhf.
(Reading database ... 18179 files and directories currently installed.)
Preparing to unpack package.deb ...
Unpacking libjpeg8:armhf (8d-1+deb7u1) ...
Setting up libjpeg8:armhf (8d-1+deb7u1) ...
 [Info] Apt-get is processing, please wait...

E: Unable to correct problems, you have held broken packages.
```

Fourdee, 2017-10-29:
Ok, lets try pulling in the missing packages for ARMv7/v8 from various sources, prior to install.

Fourdee, 2017-10-29:
Completed

closed 2017-10-29 by Fourdee

ryaneneal, 2018-01-18:
I was able to install it on Stretch with this, as su:
curl -sSL https://get.docker.com/ | sh
docker run -it --rm -v /usr/local/bin:/target emby/embyserver:armv7 instl