MichaIng/DietPi #1155: Stretch | MotionEye fails to install (deps)

Fourdee, 2017-09-22:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=986&start=30#p8818

MichaIng, 2017-09-29:
dietpi-software downloads/expects the "jessie" version here with related older library dependencies: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L6376-L6405

On official raspbian stretch repo there is "motion" available, auto installing the mentioned newer library versions as dependencies. I don't know about the python packages, but they seem to be not affected anyway and could be installed afterwards.

Fourdee, 2017-10-13:
@MichaIng 

Confirmed, RPi Stretch + x86_64 Stretch :
```
root@DietPi:~# apt-cache search motion | grep ^motion
motion - V4L capture program supporting motion detection
```

>  Mr-Dave released this on Oct 25, 2016

Also looks like the motion project has stopped devolpment:
https://github.com/Motion-Project/motion/releases

So yep, for Stretch we can install via apt.

Fourdee, 2017-10-13:
Ok, install fixed. Not tested camera yet.

Fourdee, 2017-10-14:
Putting https://github.com/Fourdee/DietPi/issues/1155#issuecomment-336500461 down to my camera failure (this being the 2nd one lol).

Also tried:
- Jessie
- ```rpi-update``` kernel

Same issues.

As installation is now resolved, marking as closed.

closed 2017-10-14 by Fourdee