MichaIng/DietPi #1156: Odroid C2 WiFi Fail

qpens, 2017-09-22:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 156
- SBC Device Odroid C2
- Power supply used 5v 4A
- SD card used Lexar
- Wifi Edimax EW-7811Un
- Distro (EG: Jessie) | Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:
After enabling and setting up wifi in dietpi.txt I should get wifi connection
#### Actual behaviour:
No IP no Connection no WiFi

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-09-23:
@qpens 

Thanks for the report 👍 

> After enabling and setting up wifi in dietpi.txt I should get wifi connection

Can you please confirm the ```dietpi.txt``` changes were made prior to 1st run of the device?
If you are applying these settings, after this point, it needs to be completed in ```dietpi-config```
https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L4-L8

And ```Wifi_Enabled=1``` was set in ```dietpi.txt```?

qpens, 2017-09-23:
lsusb result 

[image]

On second sd card, I made fresh install with that config 

[image]

[image]

qpens, 2017-09-24:
I checked another card Alfa AWUS051NH on V144 works fine but after upgrade to 156 It's a fail

[image]

qpens, 2017-09-25:
I tested both and they are working fine problem with Edimax was that this card needs to be configured with static IP on start (V144)

Alfa was working fine on V144 but after upgrade to V156 same thing happen

Fourdee, 2018-02-22:
Marking as closed due to outdated.

closed 2018-02-22 by Fourdee