MichaIng/DietPi #1158: Change login notification to reflect possibility of automatic installation

einstweilenhier, 2017-09-24:
https://github.com/Fourdee/DietPi/blob/52ee1003d41ddf1619267883676ef84210e91184/dietpi/login#L211

This line is confusing when you log into your system for the first time if you use automatic installation, because it implies that user interaction is necessary. 
Maybe it should be appended with "or wait for automatic installation to finish, if you enabled it".

Or maybe even detect if the dietpi-software instance running in the background is doing an automatic installation?

I'm sorry for not submitting a pull request but I'm not sure how any of this ties into other parts of the system.

Fourdee, 2017-09-25:
@einstweilenhier 

> Maybe it should be appended with "or wait for automatic installation to finish, if you enabled it".
Or maybe even detect if the dietpi-software instance running in the background is doing an automatic installation and just tell the user to come back later?

Many thanks for letting us know, you are spot on 👍 

I'll get this fixed.

Fourdee, 2017-10-14:
Improved for v157: https://github.com/Fourdee/DietPi/commit/c6563a27bf3e58bdf27bbc1ac28a6dba936018cc#diff-1bbeaae6dce32a61c2d5f35c3ce5c971

closed 2017-10-14 by Fourdee