MichaIng/DietPi #1159: Letsencrypt supports Free Noip.com Dynamic DNS

techdabbler, 2017-09-26:
Letsencrypt currently/now supports the free version of noip.com's DDNS service -- so it seems like we can remove the warning from dietpi.

The only thing I think that needs to be done is to change the text on the installer and on the webpage to remove the warning it doesn't work.  It might even be a good idea to just change it to it DOES work - since it has been saying otherwise.

I've successfully used dietpi-letsencrypt to grab a cert for a noip.com free ddns address so I know it works.

If I get some time I'll try to hunt down the changes and submit them but it seems it might be quicker for someone to make the changes who knows where in the code the text is.

Fourdee, 2017-09-26:
@techdabbler 

> Letsencrypt currently/now supports the free version of noip.com's DDNS service

Excellent, thanks for letting us know.

> If I get some time I'll try to hunt down the changes and submit them but it seems it might be quicker for someone to make the changes who knows where in the code the text is.

I'll add it to my list aswell.

techdabbler, 2017-09-28:
I got to digging tonight... and thinking.  I don't believe there is any code that needs changing or editing. It seems the only notation to noip.com free's service not working is just in the certbot wiki entry.

Bridouz, 2017-09-28:
From what I know Letsencrypt supports noip free accounts for a while. Always been using DietPi with free NOIP for a year :)

techdabbler, 2017-09-28:
@Bridouz Yup. It seems its been a while. I just started reading the documentation first (GASP!) and was happy to dsicover through some googling they are supported now.  I'd like to save the other 2 people who read the documentation before they try it some time. HA!

MichaIng, 2018-02-01:
Also the hint, that nginx is not supportet is obsolete. I will take care of it together with the other certbot rework: https://github.com/Fourdee/DietPi/issues/734

MichaIng, 2018-04-26:
Hmm, I think I just fixed the Nginx support hint, but can't find something about noip any more. Was this already fixed before?